meshtastic/firmware #5: support for LilyGo TTGO v1 board ?

saket424, 2020-02-28:
How hard would it be to add support for the https://github.com/LilyGO/TTGO-LORA-V1 in addition to Heltec board that you already support?

$36 for a pair of them and they have a nice 3d-printed case
https://www.aliexpress.com/item/32842155838.html

geeksville, 2020-02-28:
This would be **super** easy to add - because all of the TTGO devices are quite similar in most ways (I've used a bunch of them on previous projects).  If you wanna get a couple of these devices and add the support I'm happy to provide further pointers/help.  It should be mostly just a change to configuration.h

geeksville, 2020-02-28:
(temporarily marking you as a person who might be doing this ;-) )

saket424, 2020-02-28:
I bricked a TTGO v1 board by trying to flash the heltec image. I am unable to erase_flash anymore. Pretty stupid of me to have tried that.

I guess if you tell me exactly what changes I need to make to configuration.h for the TTGO v1 board, I am willing to attempt and sacrifice one more for this noble cause.

Thanks

saket424, 2020-02-29:
Hope this helps you in making a new configuration.h

https://github.com/espressif/arduino-esp32/blob/master/variants/ttgo-lora32-v1/pins_arduino.h

geeksville, 2020-02-29:
by coincidence it sounds like @claesg might be using this same board with success.  I've asked him to chime in.

Btw: I think it is quite possible your board is not bricked.  From looking at the docs you linked to, I didn't see anything fatally incompatible wrt the Heltec device.  I suspect that if it is seriously messed up in the appload, the only thing you need to do to unbrick it is to hold GPIO 0 to ground while powering on the board (to keep it in the bootloader), then something like "esptool.py chip_id" should be able to see your board again.

I've accidentially flashed the heltec load on my ttgo beams many times without a problem.

geeksville, 2020-02-29:
though this did show a problem ;-) : https://github.com/geeksville/Meshtastic-esp32/issues/10

claesg, 2020-02-29:
I have flashed the prebuilt 0.0.5 zipfiles to this board:
https://www.banggood.com/LILYGO-TTGO-LORA32-868Mhz-SX1276-ESP32-Oled-Display-bluetooth-WIFI-Lora-Development-Module-Board-p-1248652.html?cur_warehouse=UK
I think it is a V1 TTGO board.

saket424, 2020-03-01:
@geeksville 
Your hunch was right, what i thought a bricked esp32 came back to life today on its own

@claesg 
No such luck for me. When I tried flashing the pre-compiled 0.0.5 US, it just rolls in this

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:251564
ets Jun  8 2016 00:22:57

claesg, 2020-03-01:
@saket424 
This is what it look like when I flash the Heltec build of 0.0.5 from the zip-file

check for boards:

$ esptool.py chip_id
esptool.py v2.6
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting........___
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 3c:71:bf:f0:86:ac
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 3c:71:bf:f0:86:ac
Hard resetting via RTS pin...

flash:

$ esptool.py write_flash 0x10000 release/latest/firmware-HELTEC-EU-0.0.5.bin 
esptool.py v2.6
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 3c:71:bf:f0:86:ac
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1202416 bytes to 662970...
Wrote 1202416 bytes (662970 compressed) at 0x00010000 in 58.7 seconds (effective 163.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

geeksville, 2020-03-02:
hmm - I just looked at the pinouts and memory info at https://github.com/LilyGO/TTGO-LORA-V1

And I can't really help too much by debugging (I don't have that board type) remotely but it sure looks almost identical to the heltec board.  I'm happy to help if you have questions but I suspect some local debugging will be involved.

saket424, 2020-03-04:
@geeksville ,
For the moment I have ordered a HELTEC v2 board from https://www.amazon.com/NEXTION-Display-Transceiver-Bluetooth-Development/dp/B07916Q3T6/ref=cm_cr_arp_d_product_top?ie=UTF8 and some TTBEAMs . After I am successful with them, I will try to figure out what makes the TTGO-V1 board different

Thanks

geeksville, 2020-03-06:
ooh cool - thats good to know.  yah, I could also see how flashing to 0x1000 could hurt bootability because there is some important system stuff esp32 stores down there.

I'm reopening this issue as "we should cleanup README.md to list supported devices in a table and add this new device."  If you wanna submit a PR that's great but no pressure - we'll get to it someday.

geeksville, 2020-03-10:
README fixed by @claesg.  Thanks!

closed 2020-03-10 by geeksville