meshtastic/firmware #7: Localisation (Polish lang)

texrg, 2020-02-28:
Is possible add Polish language to project?
(and for web page)

Now exist many system to translate over webpage

geeksville, 2020-02-28:
Hi @texrg, yes, that would be a great addition.  Localization is definitely in the "eventual" todo list, but we would eagerly take pull requests if someone wants to contribute something sooner.  There are very few strings in the device side code, and a number of good possible toolchains that could be selected.  Once our strings have been made localizable (and probably a second language as the impetus for that switch) I'd promise we'd never again break that ;-)

I think until some volunteer decides this is the feature they want to work on this will go in the "things to do after the 1.0" release pile.

The sister android app also has strings and the android support for localization is even more mature, so quite easy to add there (and not too hard to add here).

geeksville, 2020-04-09:
good news! We now support localization.  Would you be willing to contribute Polish?  the info is here: https://meshtastic.discourse.group/t/call-for-help-please-localize-the-app-for-your-language/107

geeksville, 2020-06-23:
ok - no word for @txerg. closing for now.

closed 2020-06-23 by geeksville