meshtastic/firmware #11: Currently we always claim GPS is available on TTGO boards

geeksville, 2020-02-29:
Based on a comment from @claesg that he got the TBEAM load working with this: https://www.banggood.com/LILYGO-TTGO-LORA32-868Mhz-SX1276-ESP32-Oled-Display-bluetooth-WIFI-Lora-Development-Module-Board-p-1248652.html?cur_warehouse=UK

We should probably change the way we set the flag in NodeDB.cpp:
```
    myNodeInfo.has_gps = true;
```
based on board type.  But really, we should always set that flag based on "have we ever successfully talked to the GPS" - this should be done from GPS.cpp if the encode() routine ever returns true.

This would allow proper reporting of that flag to phones based on what we dynamically discovered to actually be connected on our board.  This is important, because if the phone sees that flag as false, it assumes the local device doesn't have a GPS so the phone spends battery to periodically use the android GPS to send location data for the current node.

claesg, 2020-03-01:
Well, I am using the Heltec build for the TTGO LORA32 board. I choose that one since it had no GPS and the pins had the same connections as my board.

girtsf, 2020-03-14:
I'll poke at this today.

geeksville, 2020-03-15:
fixed in https://github.com/meshtastic/Meshtastic-esp32/commit/64109b25f24b0b76fe077ba039a1d9a550561f67

commit bb9f595b (2020-05-03): Fix #11
closed 2020-03-15 by geeksville