meshtastic/firmware #13: Supported frequencies

astro-arphid, 2020-03-04:
Hi! I'm really excited to give this a go, but just want to check something before flashing.

In the UK, the use of [433MHz is also permitted](https://www.thethingsnetwork.org/docs/lorawan/frequencies-by-country.html#u), but this isn't listed in the documentation here. Given that my hardware is compatible with this band, would Meshtastic work out of the box, or would additional changes to the code need to be made to support this?

Thanks!

astro-arphid, 2020-03-04:
Sorry - I got excited and didn't fully look over everything. Checked the firmware releases and there isn't anything for the hardware requirement at this time.

Closing for now, but will look into building a working version based on the documentation.

Thanks!

geeksville, 2020-03-04:
@astro-arphid  Sounds good.  If you wanna shoot over a PR to add a EU433 region that would be great.  Essentially you'd need to

* change bin/build-all to add EU433 
* change src/MeshRadio.h (see CH0_EU for example) 
* and the README.md to mention the new choice of 433 mhz radios for EU.

geeksville, 2020-03-04:
(the current code for EU builds configures the radios for 865MHz - which you definitely don't want if the mfg stuffed parts for 433MHz)

geeksville, 2020-03-05:
btw @astro-arphid do you think you might be able to sign up for this task?  I'm sort of able to do it, but it would be better if someone that actually has 433MHz hardware tried tested it.

astro-arphid, 2020-03-06:
@geeksville I'll certainly give it a go this weekend, if everything goes to plan!

astro-arphid, 2020-03-06:
@geeksville Looking over the [LoRA Alliance paramaters](https://lora-alliance.org/resource-hub/rp2-101-lorawanr-regional-parameters) at the moment and just want to check that I'm following this correctly - let me know if this isn't the standard being worked with. 

Based on the documentation, 433.175MHz is the minimum center frequency and 434.665MHz is the maximum, where 433.175, 433.375 and 433.575 are specified as channels 0, 1 and 2. Would it make sense to have the following, maxing out at 8 channels, where channel 8 would be 434.575MHz?

```
// src/MeshRadio.h
// EU433 channel settings
#define CH0_EU443          433.175f // MHz
#define CH_SPACING_EU433   0.2f // MHz
#define NUM_CHANNELS_EU433 8
```

geeksville, 2020-03-06:
Yep.  That looks perfect!

(Sent from a phone - please ignore typos)

astro-arphid, 2020-03-07:
@geeksville I should be able to test this shortly. Building at the moment, but noticed that the board is set to `ttgo-t-beam` by PlatformIO even when making sure that `HELTEC_LORA32` is defined in configuration.h - is this okay or should I be changing this in the PlatformIO configuration also?

For example, when building within VSCode I see this:

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-beam.html
PLATFORM: Espressif 32 1.11.2 > TTGO T-Beam
HARDWARE: ESP32 240MHz, 1.25MB RAM, 4MB Flash
```

geeksville, 2020-03-07:
That's okay.  The only thing that really matters is the #define in
configuration.h.  For the sort of stuff platformio cares about the two
boards are identical (though the  Heltec has more FLASH and the TTGO has
more RAM).  Someday it would be awesome to figure out how to have the build
env stuff in platformio.ini control our build (instead of crufty changing
those #defs).  It shouldn't be that hard (you can see my early attempts at
the end of platformio.ini, but I just couldn't get it working.

geeksville, 2020-03-10:
@astro-arphid no pressure but just checking in: did the 440mhz stuff work for you?

astro-arphid, 2020-03-10:
@geeksville Sorry, I'll try and get that tonight or tomorrow. I've been able to get the two to pair on separate devices and send messages to each other. I had an initial issue where after a few minutes, bluetooth would disable on both of them and I had to power cycle them, before the same thing would happen again.

It looks like this was picked up based on the gitter chat, I pulled the newer version down and uploaded to the radios, but haven't had the chance to properly test - just a quick go at it the last couple of mornings. I might be the OnePlus user mentioned on the google group.

spattinson, 2020-03-11:
Im interested in 433Mhz well 434 really, I think it gets you out of the commercial band of door bells, garage openers, etc. I have a couple of ttgo tbeams I'd like to try this out.. looks amazing from what I have seen so far.

closed 2020-03-12 by geeksville