meshtastic/firmware #14: Make an iOS client

geeksville, 2020-03-06:
Hi @kellyp (cc @girtsf),

Kelly was kind enough to be interested in making the iOS client (woot!).  So I just made some notes on how the bluetooth API works.  These notes might also be handy for Girts' interest in making a protocol client across the serial port (or possibly instead make a python bluetooth client that uses the bluetooth API? people might really dig that):

https://github.com/meshtastic/Meshtastic-esp32/blob/master/docs/software/bluetooth-api.md#meshbluetoothservice

geeksville, 2020-03-11:
Hi @kellyp, Meet @jeksys (Eugene),

Eugene is **also** interested in making an iOS for this project (and omg we really need one ;-)).  Perhaps you two can talk here and work out your various interests and time commitments and decide what you can sign up to do? 

I used to work at Apple many years ago, but that was pre iOS so I know nothing about such things ;-).

geeksville, 2020-03-11:
copypasta from my original chat with @kellyp (because useful context):

> That is super super cool. Historically I've done most of the development on the android app and the device code. A super great person named @girtsf stepped up and hopefully he will soon be driving the device code and I'll focus on the android app. We'd love to have a compatriot that makes an iOS app happen. The bluetooth API to the device was structured intentionally to minimize the amount of work phone app writers will need to spend. For instance, the android app is essentially four screens (messages, channel settings/qr code, user settings/bluetooth pairing, and the list of nodes in the mesh with their distance and heading). Behind the scenes the android app has a small service that just talks bluetooth to the radio (RadioInterfaceService) and a higher level service that knows about the mesh that uses that service (MeshService). We have some decent docs on the bluetooth API and the protocol buffers we use for transport.

jeksys, 2020-03-11:
@geeksville can you please share screenshots from the Android app. And if you can create a video how the android app works that would help as well. 

I also created Slack workspace for devs to chat: **meshtastic.slack.com**

seclorum, 2020-03-11:
Just wanna check in and say I can help with iOS dev .. got tons of experience with iOS, including low-level Bluetooth .. hit me up if any help is needed - happy to assist in any capacity, testing, etc.

geeksville, 2020-03-11:
Hi @jeksys - happily!  I'll grab some screenshots tomorrow and post them here.  I can also post instructions on how to run the standard (free) android dev emulator, so if you wanna play with the live app you could do that also (though when it detects it is under an emulator it only talks a sim mesh of three devices).

geeksville, 2020-03-12:
@jeksys Here's some fresh screenshots for your noble effort.  If you could take this up it would be awesome: https://github.com/meshtastic/Meshtastic-Android/commit/4f1d85529b6ebed4ee6849746301fb6799201418

I'll make a video before the end of the week and also provide instructions on how to run the android emulator.

jeksys, 2020-03-13:
@geeksville thank you for screenshots. Can you please create a new Meshtastic-iOS repo

@seclorum is there any open source BLE library you can recommend to use for the project?

geeksville, 2020-03-13:
@jeksys will do now, also I'll invite you to the github group.  see your email.  yay!

seclorum, 2020-03-14:
@jeksys : key thing here is, KISS.  On iOS - do not introduce 3rd-party complexities to the issue, instead use Apples' CBCentralManager (CoreBluetooth) to manage all BTLE devices.  It will do what you need.  If you get stuck, let me know - I'm happy to assist with debugging/implementation details as you proceed.

seclorum, 2020-03-14:
This is a pretty good introduction to the subject:

https://codeburst.io/getting-started-with-bluetooth-low-energy-on-ios-ada3090fc9cc

I would not suggest using Swift, however.  At least, not yet.

geeksville, 2020-03-15:
sorry @jeksys - I lied again: https://meshtastic.discourse.group/t/device-code-release-announcements-0-1-6/32

seclorum, 2020-03-17:
@jeksys Check out this great reference that might come in quite handy for your efforts:

https://littlegreenviper.com/series/bluetooth/

jeksys, 2020-03-17:
@seclorum thank you the link. I've done BLE device integration before. Will definitely ask for your help if hit any road blocks. Waiting for hardware for now and slowly starting work on the app

geeksville, 2020-03-24:
@jeksys I'm sending you an invite now to join the devgroup.  I think it will show up in your mail and you'll need to opt-in.  I've been away for sixish days sick.

geeksville, 2020-04-11:
@jeksys I'd like to turn on the "require all org members to have 2FA enabled for their github account" feature (to prevent some accident from having someone come in and nuke code).  Are you okay with that? (github will pester you to turn on this feature)

jeksys, 2020-04-11:
@geeksville I’m ok with that.

Finally got my device,  so I can start building the app

geeksville, 2020-04-11:
@jeksys awesome!  feel free to ping me any time if you have questions (and see the sw docs/software directory).  Also if @girtsf proposal to simplify the bluetooth API would help you I can apply that change now (to decrease the amount of work you face).

geeksville, 2020-04-11:
Also - I wouldn't recommend adding a map in your initial release.  But if/when you do want to add a map, possibly use Mapbox.  I used them on the android app (and a previous commercial project), and their docs are great, the free usage tier is generous enough for this sort of project, and if the iOS API is similar to their Android API you could crib from my MapFragment.kt.

jeksys, 2020-04-11:
@geeksville can you please add me back to the organization. I enabled 2FA.

Sure, the simpler the better and faster to implement

geeksville, 2020-04-28:
@jeksys just checking in, any progress on your grand adventure?  (no pressure - but curious)

Also, I finally implemented a (fairly nice IMO) python API and test client based on an idea from @girtsf.  If you a super minimal example of a device client to refer to you will probably prefer reading just a few pages of python than the huge ball of 'things you need for Android' in the android app.  

Here it is:

https://pypi.org/project/meshtastic/

If you run it against a current device load I think just running the "meshtastic" command line app should let you see interesting things.

jeksys, 2020-04-28:
@geeksville 
https://pypi.org/project/meshtastic it's actually a good suggestion. I'm also using existing docs and android app. 

I've started working on a prototype and connecting to my board. Next step is to pull all together and build the first - "connection view" of the app. Once it's ready I will submit the code. 

You mentioned that there is going to be a new BLE protocol for the board, are you going to release it anytime soon?

geeksville, 2020-04-28:
@jeksys ooh that's awesome and way better than I was guessing.

Yes, the new API is in the device software I just released (0.6.0).  It is backwards compatible, so you can use the old or new API to talk to it.  In RadioInterface.kt of the android app, you'll see a temporary bool called "useOldApi" which probes to see if the device can understand the old messages.  You can change that to "false" and it will only use the new API.

Also, the little python thing I just made only uses the new API (which like Girts proposed really is easier to use).  

The flow for a BLE client is:
* Write a ToRadio.want_config_id packet to toradio
* Then repeatedly read from fromradio until you get back an empty response.  Those responses will be my_info, radio, and then a series of node_infos (one per node in the node DB).  Use thse node infos to build your own 'nodeDB'.
* Subscribe for BLE notifies on fromnum.  If you receive a fromnum notification that means there is new data waiting for you in fromradio.
* Any time you want to send a message or change radio state, write a suitable ToRadio to the toradio characteristic.

geeksville, 2020-06-06:
Hi @jeksys and @kellyp,

So (for personal interest) I'm interested in playing with flutter.  If I were to make a **very** bare flutter (or some other cross platform kit?) version of a meshtastic client, would that be of use to your efforts?  

i.e. I could make the initial framework, use it to send/show messages and crude position info.  Build and run it on linux and android.  But - even though I worked for Apple for a number of years (pre iOS days) - I know nothing about iOS and I don't have a mac.  Would you be able to take 'ownership' of a flutter client and build/release it for iOS?

(just an idea)

jeksys, 2020-06-07:
Hi Kevin, that works.
I was actually thinking to use it as well. I can help with iOS part if you
want to start building it.
I finally got a new device, my old one died. Hopefully will start working
on standalone app as well.

geeksville, 2020-06-07:
Sounds good - if I end up playing with this (late next week or the week after?) I'll ping you so we can stay in sync.

macflyhub, 2020-08-06:
Really need to get the iOS (iPhone) Meshtastic party started..
There are a few names in this thread but not seeing much action - is there another iOS dev thread I am missing?
I am a PHP, DB and C++ coder but also help out with plenty of devs and alpha testing.
We need to rumble this up. What can I do to make this happen faster, short of buy a mac and do it myself?
Who is actively bashing away at getting this pre-started before waiting for 1.0
iOS SOS!

jeksys, 2020-08-07:
> We need to rumble this up. What cans  I do to make this happen faster,

I’m playing with some BLE prototype iOS code.
Afaik there is also another group working on the iOS client as well.
I’ll try to be more active

geeksville, 2020-08-10:
I think @grcasanova was working on something.  Alas, I took two weeks off and only now getting back at coding.  I don't currently have a Mac so I can't help with this one (though I did work in Apple kernel engineering - that was a while ago)

grcasanova, 2020-08-10:
> I’m playing with some BLE prototype iOS code. Afaik there is also another group working on the iOS client as well. I’ll try to be more active
> -- Sent from Gmail Mobile

Hi yes I'm currently developing the iOS app, stuck on BLE

geeksville, 2020-08-12:
@grcasanova If you want I can try your project on android.  Is the build procedure to:

1. Install ionic per their website (I haven't looked yet)
2. clone https://github.com/grcasanova/Meshtastic-app
3. Build for android per the ionic instructions

? Anything else I should need?  Would that help?

grcasanova, 2020-08-13:
you need to install capacitor and run npm install to fetch all plugins

geeksville, 2020-08-13:
@grcasanova ok - sounds good!  I'll play with this later today.

macflyhub, 2020-08-13:
The world is watching and waiting... no pressure!

magicwave, 2020-09-19:
I am interested to help with the iOS effort. I have some years of hospital enterprise and secure communications app development, though no experience with BLE. I'm not sure why an earlier poster recommended against Swift language but I have experience in Swift and Objective C, and also with shared C/C++ libraries between iOS and Android code bases. 

A little over a week ago my area of southern Oregon was devastated by fires. Many whose homes were lost could have evacuated with more of their belongings had there been a better notification system. I found this project while trying to imagine a solution. At least I'd love to be part of the iOS conversation. Thanks.

jeksys, 2020-09-19:
Hi @magicwave 
It's sad what happened on West Coast this year. I hope you and your family is ok. 

There is nothing against Swift. On the contrary initially I wanted to use Swift and SwiftUI for the app. But I didn't have enough time during the summer to make any significant progress, but prototyping. 

Afaik there was another group building something. 
Fall is coming, so I hope to have more time for this project. 
Let's coordinate and may be finally build iOS client. 

We need at least two components for the app: 
- Simple UI;
- BLE communication manager.

As the first step do you want to have a look at the existing Android app, BLE protocol? 
Next week I will continue my work on the BLE part.

magicwave, 2020-09-19:
@jeksys Thanks. We're ok; many near us are not so much.
Ok, great. I just stumbled on this project so I'll have a look at the Android app and keep in touch. Thanks.

geeksville, 2020-09-19:
Hi @jeksys and @magicwave whatever ya'll want to do sounds awesome.

Alas - I think that other effort fizzled because the Ionic BLE libs were pretty buggy.

So I think the remaining options are a swift app or perhaps Flutter.

There is now a decent doc on the protocol: 
https://github.com/meshtastic/Meshtastic-device/blob/master/docs/software/device-api.md

Also, if interested in a reference implementation the python API/tool might be a better first reference - because it is much simpler/smaller.  The API on bluetooth vs over serial is almost identical (see the md file above for why)

I'm happy to help if you have any questions.

magicwave, 2020-09-20:
@geeksville - thanks for the support. It would be good to know if there will be an effort to implement in Flutter. I have only the most beginning experimental development experience with Flutter, and I would be happy to help with that to the extent of my time/experience, but if it could do the job then the obvious advantage is that it is a single code base for both Android/iOS platforms.

If this were to be considered the preferred direction, iOS development would simply be duplicating efforts.

antoineclaval, 2021-02-01:
Hi 
Unfortunately, an IOS client is sorely needed, but I don't read any blocking issues here. Awesome! 

I would gladly write test or do menial tasks to help ( I don't have the IOS skills to drive, but I could mess with flutters/swift as needed ) 
A basic way to read/send messages on IOS devices would complete the use case I have in mind for meshtastic. Exciting! 

2 questions: 

Is there is any branch / repo to look at or play with regarding an IOS client? 
Is that doc still relevant to meshtastic regarding the BLE integration ? 
https://codeburst.io/getting-started-with-bluetooth-low-energy-on-ios-ada3090fc9cc

All that is pretty exciting!

sachaw, 2021-02-01:
React native or nativescript may be an option and just use out JavaScript library

cjpais, 2021-02-01:
I am interested in contributing to iOS development, I have some experience with Swift and SwiftUI. I've not got a meshtastic device in my hands yet, however plan on getting at least two for testing purposes. 

I've got the same question as @antoineclaval, regarding existing repo. If there's not I don't mind spinning one up and getting some basic work going shortly.

As @sachaw mentioned can also go react way, I've got some experience there as well and can contribute.

I am not super familiar with Bluetooh/BLE/CoreBluetooth, however I think I should be able to work out details. I formerly worked on 5G SW.

jeksys, 2021-02-01:
@antoineclaval 
I have some unfinished code with BLE manager and protobuf implementation. This code is 3-4 months old. I will push it into the iOS repo soon.

@cjpais any help is appreciated. May be with your help I can move it from the dead stop.

thegcon, 2021-02-04:
Hello all, I read through what you guys have been working on in regards to the IOS app. Does anyone have a link to where I could install the latest (non-app store) IOS alpha/beta app?  

I plan to start selling the TTGO v1.1 boards in different packages and would like to help get a working IOS app going. I can't offer any coding skills but I can offer some monetary incentives if it would help get someone supper motivated to get it finished and released openly quickly.

jeksys, 2021-02-04:
@thegcon Hi, there are no alpha build unfortunately. Unless some one else is working on the app in parallel. 
I've started a project, but haven't finished it to the release yet. 

I'm going to review my latest code and submit it soon

thegcon, 2021-02-04:
Sounds good.; appreciate your efforts. This is a really cool project and it will be great to see an IOS app for it and open it up for a lot more people.

 How far out do you expect a similar featured ios app to be? Do you know anyone else who could help but might need a monetary incentive to help?

jeksys, 2021-02-04:
Let me try to build Android app to see what's needed in term of feature and UI. 

I also want to see if my current code works with the latest firmware. 
I hope that once I push something into the repo people will start helping.

jneuf, 2021-02-12:
I'm ready to help contribute once some code gets pushed.  I dont' want to start from scratch if other folks have had some success.  Keen to help.

jeksys, 2021-02-12:
Sounds good.
Let me figure it out one last blocker - I want to fix device init and read
node info and I will push the code.

Are you comfortable with SwiftUI?

jneuf, 2021-02-12:
> Sounds good. Let me figure it out one last blocker - I want to fix device init and read node info and I will push the code. Are you comfortable with SwiftUI?
> -- Sent from Gmail Mobile

I'm not an expert but I can do basic stuff.  I've done a little bit of just simple apps for myself.

jeksys, 2021-02-12:
I’m learning myself.

sticilface, 2021-02-13:
New to the lib today!  I've just obtained 2 Tbeams, got two displays I have lying around working with `USE_SH1106` and bingo its working!    The only snag is I have an iPhone and my partner has an android phone.  So she is set.  

I have no experience with iOS development but it is on my list.  

I do have lots of hobby experience with c++, arduino, ESP8266 and ESP32.  I am very excited by this project and I am totally set on helping develop the embedded side of things.  I basically had this idea a year or two ago, bought some LORA radios, wired them up but things just got in the way.  

I would be keen to help test and development iOS apps in the works.

dkarp0, 2021-02-14:
Likewise, I'm a professional software engineer and am in need an iOS version. So will be happy to help once you have a first version pushed.

tcote19, 2021-03-19:
Hey @geeksville, @jeksys and everyone else.

Came across the meshtastic project a couple months back and picked up a few TTGO devices to play around with. I'm very interested in helping with the iOS app work too. I'm new to Swift coding but picking it up well and have a good amount of free time (currently working on a side venture with few deadlines).

At the least, I can help navigate the App Store process. I used to work within the infamous Apple App Store Review team so can help with any "gotchas" or reach out to friends of mine who are still there for questions.

I haven't been coding much lately but am eager to ramp up on iOS development.

jeksys, 2021-03-19:
@geeksville can you please add me as a contributor to Meshtastic-iOS project. 
I want to push what I have at the moment. I hope with all the help we can make it happen.

antoineclaval, 2021-03-22:
@jeksys : I just want to re-iterate my interest in IOS client,  messing with unpolished code in a branch is fine with me. 
Please tag me and I will try to be helpful. ( I would happily follow direction if you need something specific done )

geeksville, 2021-03-23:
hi ya'll, perhaps work together with @thepoweroftwo on this?  He's made great progress on an iOS client:

https://meshtastic.discourse.group/t/meshtastic-ios-app-first-alpha-release/2733

Perhaps we could make that client meshtastic-iOS and I could just delete the old used repo?

jeksys, 2021-03-23:
@geeksville that's cool. Will join his project. 

my unfinished project if any one is interested:
https://github.com/jeksys/Meshtastic-iOS