meshtastic/firmware #15: TBEAM stuck in loop

geeksville, 2020-03-08:
From @claesg  

TTGO T-BEAN stuck in boot loop.

Yesterday it was running fine (on 0.1.0) and this morning it was dead. I connected it to arduino ide and the serial monitor and noticed that it was looping:
k⸮⸮)Setting GPS power=0
ets Jun 8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
booted, wake cause 4 (boot count 6)
I2C device found at address 0x34
axp192 PMU found
I2C device found at address 0x3c
ssd1306 display found
done
chip id detect 0x3
Detect CHIP :AXP192
OUTPUT Register 0x57
AXP192 Begin PASS
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: DISABLE
DCDC3: ENABLE

Exten: ENABLE
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
SRC REG:0xc4
Charging enable is enable
Charging target-voltage : 0x2
end when the charge current is lower than 10% of the set value
Charge current : 700.00 mA
enable [0x40]val:0xdc
enable [0x41]val:0xff
Meshtastic swver=0.1.0, hwver=1.0-EU
Read RTC time as 21 (cur millis 1503) valid=0
Screen: Started...
Loading saved preferences
NODENUM=0xa0, dbsize=1
Starting meshradio init...
IRQ flag mask 0x0
Set radio: name=. config=0, ch=0, txpower=0
Trigger powerFSM 6
Transition powerFSM transition=Boot, from=DARK to=ON
Setting bluetooth enable=1
Pre BT: 187784 heap size
Starting bluetooth
* Mesh service:
Transition powerFSM transition=Screen-on timeout, from=ON to=DARK
Transition powerFSM transition=Phone timeout, from=DARK to=NB
Setting bluetooth enable=0
Shutdown BT: 186980 heap size
Transition powerFSM transition=Min wake timeout, from=NB to=LS
Transition powerFSM transition=mesh timeout, from=LS to=SDS
Setting GPS power=1
Looking for GPS lock
Entering deep sleep for 0 seconds
Writing preferences
Setting GPS power=0
ets Jun 8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
booted, wake cause 4 (boot count 7)
I2C device found at address 0x34
axp192 PMU found
I2C device found at address 0x3c
ssd1306 display found
done
chip id detect 0x3
Detect CHIP :AXP192
OUTPUT Register 0x57
AXP192 Begin PASS
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: DISABLE
DCDC3: ENABLE

Exten: ENABLE
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
SRC REG:0xc4
Charging enable is enable
Charging target-voltage : 0x2
end when the charge current is lower than 10% of the set value
Charge current : 700.00 mA
enable [0x40]val:0xdc
enable [0x41]val:0xff
Meshtastic swver=0.1.0, hwver=1.0-EU
Read RTC time as 25 (cur millis 1503) valid=0
Screen: Started...
Loading saved preferences
NODENUM=0xa0, dbsize=1
Starting meshradio init...
IRQ flag mask 0x0
Set radio: name=. config=0, ch=0, txpower=0
Trigger powerFSM 6
Transition powerFSM transition=Boot, from=DARK to=ON
Setting bluetooth enable=1
Pre BT: 187784 heap size
Starting bluetooth
* Mesh service:
Transition powerFSM transition=Screen-on timeout, from=ON to=DARK
Transition powerFSM transition=Phone timeout, from=DARK to=NB

Setting bluetooth enable=0
I tried to reflash meshtastic but it did not help. I had to flash 'wifi scan' example from the ide which it accepted and it was working. I then flashed meshtastic and this time it is working as expected. Maybe the log above can tell what happened?

Sorry about the strange formatting.

geeksville, 2020-03-08:
Super useful.  Was it repeating this in a loop at high speed, rebooting rapidly, or was it pausing for a long time while in deep sleep.

geeksville, 2020-03-08:
oh - I see the problem:

> Entering deep sleep for 0 seconds

That number should never be zero.  I'll check the code and fix this later today.

geeksville, 2020-03-08:
ooh - yeah - I added the deep sleep parameter to the preferences we store in flash, but didn't think about that people who had installed older builds would get zeros for that value.  The fix will be easy.

claesg, 2020-03-08:
That is great!

commit 011864a2 (2020-03-08): Fix #15 Force all devices to update preferences
commit 5dbacbdd (2020-03-08): version 0.1.13 Fix #15 found by @claesg. Thanks for the detailed report!
closed 2020-03-08 by geeksville