meshtastic/firmware #17: Heltec Lora32 is not really the same as ttgo lora32

geeksville, 2020-03-08:
theory, it is the flash partitioning.

> I think I have a pretty good theory on what is wrong with programming your device (I just reproed it on a similar device of mine).  Some kind soul reported that the HELTEC image works fine on the TTGO LoRa32, but I think that somewhat depends on luck because I think TTGO and Heltec partition their flash somewhat differently.  I'll need to make a small script to let you program your device reliably.  I'll do that tonight.

geeksville, 2020-03-08:
cc @bcanderson

BCAnderson, 2020-03-09:
I have to apologize but I think building this is beyond my capabilities.  looks like I'm no help there, but I'd be happy to try whatever you like on my board.

geeksville, 2020-03-09:
no worries. I'll work on this tonight/tomorrow.

geeksville, 2020-03-09:
Hi BC,

Can you try running the following three commands on your board (using the binary file I'm attaching)?  This will allow me to test my theory.  After running these commands, power cycle the board and see if it boots?

Note: the two files you flash have different starting addresses

```
kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$ esptool.py --baud 921600 erase_flash
esptool.py v2.6
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:b5:36:7c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 16.2s
Hard resetting via RTS pin...
```

```
kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$ esptool.py --baud 921600 write_flash 0x1000 images/system-info.bin 
esptool.py v2.6
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting......
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:b5:36:7c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Compressed 61440 bytes to 11950...
Wrote 61440 bytes (11950 compressed) at 0x00001000 in 0.1 seconds (effective 3569.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

You should use the correct file for your region though...
```
kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$ esptool.py --baud 921600 write_flash 0x10000 release/latest/firmware-HELTEC-US-0.1.3.bin 
esptool.py v2.6
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:b5:36:7c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Compressed 1202448 bytes to 663131...
Wrote 1202448 bytes (663131 compressed) at 0x00010000 in 10.7 seconds (effective 900.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```
[system-info.zip](https://github.com/meshtastic/Meshtastic-esp32/files/4304204/system-info.zip)

claesg, 2020-03-10:
Hi, Just let me know if you want me to run the same thing on my TTGO Lora32 which is the one that can run the Heltec build.

illperipherals, 2020-03-16:
Nice, that worked on my TTGO LoRa boards that came in today. Was otherwise going into a crash loop with: 
```rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40084137  PS      : 0x00060e30  A0      : 0x80085ee5  A1      : 0x3ffbbd30
A2      : 0x3ffbbd58  A3      : 0xbde04382  A4      : 0xfdd04382  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00013ffc  A9      : 0x3ffbbd10
A10     : 0x00000000  A11     : 0x3ff42000  A12     : 0x00000040  A13     : 0x3ff420c0
A14     : 0x3ffba7a8  A15     : 0x00000004  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c
EXCVADDR: 0xfdd04382  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x40084137:0x3ffbbd30 0x40085ee2:0x3ffbbd50 0x400e7699:0x3ffbbd80 0x400e4078:0x3ffbbda0 0x400e3fad:0x3ffbbdc0 0x400e4b67:0x3ffbbde0 0x400e729a:0x3ffbbe10 0x400913c9:0x3ffbbe30

Rebooting...
ets Jun  8 2016 00:22:57```

BCAnderson, 2020-03-16:
"> Nice, that worked on my TTGO LoRa boards that came in today. Was otherwise going into a crash loop with:"

My TTGO LoRa board is stuck in the same (or similar) crash loop.  What did you do??

illperipherals, 2020-03-16:
I just ran the three commands above with the bin that is attached.
```
esptool.py --port COM10 --baud 921600 erase_flash
esptool.py --port COM10 --baud 921600 write_flash 0x1000 system-info.bin
esptool.py --port COM10 --baud 921600 write_flash 0x10000 firmware-HELTEC-US-0.1.6.bin

BCAnderson, 2020-03-16:
I did this:
esptool.py --port /dev/ttyUSB0 --baud 921600 erase_flash
esptool.py --port /dev/ttyUSB0 --baud 921600 write_flash 0x1000 system-info.bin
esptool.py --port /dev/ttyUSB0 --baud 921600 write_flash 0x10000 firmware-HELTEC-US-0.1.6.bin
and got this:

"ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40153630: 74010404 51010444 50511501
Core 0 register dump:
PC      : 0x40153634  PS      : 0x00050030  A0      : 0x800e5bbe  A1      : 0x3ffbbdc0  
A2      : 0x8001f880  A3      : 0x8001f880  A4      : 0x400fe632  A5      : 0xbffa2b3e  
A6      : 0x00060c20  A7      : 0x00000000  A8      : 0x800e500c  A9      : 0x3ffbbda0  
A10     : 0x15401540  A11     : 0x00000001  A12     : 0x3ffd1589  A13     : 0x00000001  
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40153634:0x3ffbbdc0 0x400e5bbb:0x3ffbbde0 0x400e82ee:0x3ffbbe10 0x400913c9:0x3ffbbe30

Rebooting..."

So no change for me. :(

geeksville, 2020-03-17:
@BCAnderson later today I'll push out a 0.1.7 build that takes advantage of a new stacktrace feature girts added.  If you try running that load and give back the stacktrace in this bug we might be able to remote debug your board from here.

BCAnderson, 2020-03-18:
Howdy, I tried 1.7 
esptool.py --port /dev/ttyUSB0 --baud 921600 erase_flash
esptool.py --port /dev/ttyUSB0 --baud 921600 write_flash 0x1000 system-info.bin
esptool.py --port /dev/ttyUSB0 --baud 921600 write_flash 0x10000 firmware-HELTEC-US-0.1.7.bin

And I got this:

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Core 0 register dump:
PC      : 0x55445400  PS      : 0x00060030  A0      : 0x800e5ba8  A1      : 0x3ffbbf20  
A2      : 0x40111450  A3      : 0x00000001  A4      : 0x3ffd20a4  A5      : 0x00000001  
A6      : 0xffffffff  A7      : 0x00000000  A8      : 0x800ff1f5  A9      : 0x3ffbbf00  
A10     : 0x3ffba3d8  A11     : 0x00000000  A12     : 0xffffffff  A13     : 0x00000000  
A14     : 0x0000001f  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x55445400:0x3ffbbf20 0x400e5ba5:0x3ffbbf40 0x400e6757:0x3ffbbf60 0x400e8e8a:0x3ffbbf90 0x400913c9:0x3ffbbfb0

Rebooting...

It's a different boot error but it still continually reboots.
How do I use the tool to capture the system portion of -flash?

geeksville, 2020-03-19:
@BCAnderson you shouldn't need to run that tool, because I grabbed the image here for you to use.

Alas, I think I'll need to come back to this  tomorrow  because (as unlikely as it sounds) I think I might actually be slightly coronamotized - and I'm going to go lay down.

BCAnderson, 2020-03-20:
Howdy Kevin
no change with  0.1.8

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400ff646  PS      : 0x00060030  A0      : 0x800e6020  A1      : 0x3ffbc080  
A2      : 0x15404101  A3      : 0x00000001  A4      : 0x05015500  A5      : 0x00000001  
A6      : 0xffffffff  A7      : 0x00000000  A8      : 0x800e6104  A9      : 0x3ffbc060  
A10     : 0x00000011  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffbd908  
A14     : 0x0000001f  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x05015500  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400ff646:0x3ffbc080 0x400e601d:0x3ffbc0a0 0x400e6bcf:0x3ffbc0c0 0x400e9302:0x3ffbc0f0 0x400913c9:0x3ffbc110

Rebooting...

geeksville, 2020-03-24:
@BCAnderson okay - I'm back from a six day flu/covid break.  I'll check this stack trace and comment later today.

geeksville, 2020-03-24:
@BCAnderson sorry to ask this, but can I ask you to try this one more time with 0.1.9 and send in the stack trace?  I would do it myself, but I don't have a board from your mfg.

for some reason, just using the git tags is not allowing me to generate a build that makes sense for that stack trace.  For instance, for that trace I get:

```
kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$ ./bin/exception_decoder.py /tmp/stack 
stack:
0x400d8df1: OLEDDisplay::getStringWidth(char const*, unsigned short) at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/esp32/ESP8266_SSD1306/src/OLEDDisplay.cpp:813
0x400d973f: OLEDDisplayUi::tick() at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/esp32/ESP8266_SSD1306/src/OLEDDisplayUi.cpp:219 (discriminator 4)
0x400d97e1: OLEDDisplayUi::update() at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/esp32/ESP8266_SSD1306/src/OLEDDisplayUi.cpp:219 (discriminator 4)
0x400d64f5: meshtastic::drawNodeInfo(OLEDDisplay*, OLEDDisplayUiState*, short, short) at /home/kevinh/development/meshtastic/meshtastic-esp32/src/screen.cpp:337
0x400d5442: initBluetooth() at /home/kevinh/development/meshtastic/meshtastic-esp32/src/main.cpp:277 (discriminator 8)
0x400e735b: _Unwind_ForcedUnwind at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind.inc:209
0x400913c9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

So I'm now doing what I should have done all along, I'm including the elf file (which contains symbols with all builds, so in the future we can always generate the correct stack trace:

https://github.com/geeksville/Meshtastic-esp32/commit/3a756b0e088bb8a31769a01b32052f2482385b1c

geeksville, 2020-03-24:
installer script item moved to https://github.com/meshtastic/Meshtastic-esp32/issues/27

BCAnderson, 2020-03-25:
Welcome back Kevin

No change with 0.1.9

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400ff756  PS      : 0x00060030  A0      : 0x800e6130  A1      : 0x3ffbc080  
A2      : 0x10000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0xffffffff  A7      : 0x00000000  A8      : 0x800e6214  A9      : 0x3ffbc060  
A10     : 0x00000011  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffbd908  
A14     : 0x0000001f  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400ff756:0x3ffbc080 0x400e612d:0x3ffbc0a0 0x400e6cdf:0x3ffbc0c0 0x400e9412:0x3ffbc0f0 0x400913c9:0x3ffbc110

Rebooting...

BCAnderson, 2020-03-25:
I apologize for my ignorance, but how do I get the stack trace?

geeksville, 2020-03-25:
re: stacktrace
No worries, that was the "backtrace" line you just provided.  And wow.  I thought I just had a mismatch of my elf file compared to what was failing on your clone board.  And nope, that really was the failure.  I'm currently thinking about what this means.

```
kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$ ./bin/exception_decoder.py -e release/latest/firmware-HELTEC-US-0.1.9.elf /tmp/stack 
stack:
0x400d8df1: OLEDDisplay::drawInternal(short, short, short, short, unsigned char const*, unsigned short, unsigned short) at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/esp32/ESP8266_SSD1306/src/OLEDDisplay.cpp:813
  \-> inlined by: OLEDDisplay::drawStringInternal(short, short, char*, unsigned short, unsigned short) at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/esp32/ESP8266_SSD1306/src/OLEDDisplay.cpp:514
0x400d973f: OLEDDisplayUi::drawIndicator() at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/esp32/ESP8266_SSD1306/src/OLEDDisplayUi.cpp:219 (discriminator 4)
0x400d97e1: OLEDDisplayUi::drawIndicator() at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/esp32/ESP8266_SSD1306/src/OLEDDisplayUi.cpp:219 (discriminator 4)
0x400d64f5: meshtastic::drawNodeInfo(OLEDDisplay*, OLEDDisplayUiState*, short, short) at /home/kevinh/development/meshtastic/meshtastic-esp32/src/screen.cpp:216
  \-> inlined by: drawNodeInfo at /home/kevinh/development/meshtastic/meshtastic-esp32/src/screen.cpp:359
0x400d5442: setup() at /home/kevinh/development/meshtastic/meshtastic-esp32/src/main.cpp:213
0x400e735b: _Unwind_FindEnclosingFunction at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/unwind-dw2-xtensa.c:193
0x400913c9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

girtsf, 2020-03-25:
Maybe the nodeinfo changes from underneath it as it is drawing?

geeksville, 2020-03-25:
@girtsf yeah, but every time and this occurs early in the boot, before printing anything else.

geeksville, 2020-03-25:
actually never mind.  There must be something wrong with how I'm running exception_decoder.  I just used a completely different backtrace and got the same output.  Looking...

geeksville, 2020-03-25:
yeah - I was dumb.  fixing.

geeksville, 2020-03-25:
better - this stack trace makes sense.  Though it seems like a totally normal startup except we blew up the first time calling the esp library.   Puzzling...  I'll think about this later today.

```
kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$ ./bin/exception_decoder.py -e release/latest/firmware-HELTEC-US-0.1.9.elf ~/tmp/stack 
stack:
0x400ff756: esp_log_level_set at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/log/log.c:125
0x400e612d: initArduino at /home/kevinh/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:202
0x400e6cdf: app_main at /home/kevinh/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:26
0x400e9412: main_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:506
0x400913c9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$
```

closed 2020-04-13 by geeksville

Gotfrogs, 2020-06-26:
I am not a programmer but did manage to complete a project using two Heltec WiFi LoRa 32 (V2) boards and sample code from online. I would like to make more of them but also save some cash. Can I use my same code on the TTGO LORA32 version of the board?

geeksville, 2020-06-28:
@Gotfrogs sorry - I didn't notice this comment until just now (for questions like this, probably faster response to post in the development forum on meshtastic.discourse.group)

Are you asking about meshtastic?  If so, it already runs on either the Heltec Lora32 or the TTGO lora32 (there are multiple build targets in our platformio.ini and we generated prebuilt bins for both of those boards).  If you are asking about your own project, the two boards are almost the same - but a few gpios are different.  See our configuration.h