meshtastic/firmware #20: TBEAM shipping currently not available to US

SeaDude, 2020-03-11:
Interesting...
I just went to purchase a TBEAM unit from your link to Ali Express. Selected the 915mHz unit. Received this message:
"Unable to ship to the US". 
Not sure if its COVID-related or another issue. Might be good to find multiple sources of these products if possible.

geeksville, 2020-03-11:
Hmm.  Interesting.  Here's a seller on amazon (though the ship date is way out - probably Covid) and the price is a bit high: https://www.amazon.com/TTGO-T-Beam-Wireless-Bluetooth-Battery/dp/B07WVWCMKZ/ref=sr_1_13?keywords=ttgo+t-beam&qid=1583893272&sr=8-13

I've had five on order through alibaba since before CNY.  I think they (along with some other aliexpress items I've ordered from other vendors) are still having a really tough time because of Covid.   I bet their backlog has now gotten so large the placed a temporary hold on new orders.

I've exchanged emails with them a few times and they seem really nice and this current situation is personally hard for them.  After they saw the hackaday article they were kind enough to airship me a few units (and some new 'watch shaped' lora gps devices) as a special favor (because I really need more than the small number I have.  Mostly so I can give two to @girtsf ;-) ).   I'm supposed to get them this week.

I'll email my contact there who was so helpful and ask if they know what's up with aliexpress orders.

SeaDude, 2020-03-11:
Thanks for the quick response! Nice work with the project. Hopefully I can get my hands on some hardware and participate.

closed 2020-03-11 by SeaDude

geeksville, 2020-03-11:
apparently Banggood has plenty of stock and they can ship pretty fast: https://www.banggood.com/LILYGO-TTGO-T-Beam-ESP32-433868915923Mhz-WiFi-Wireless-bluetooth-Module-GPS-NEO-6M-SMA-LORA32-18650-Battery-Holder-With-OLED-p-1545070.html?rmmds=search&ID=510804&cur_warehouse=CN