meshtastic/firmware #26: Support screen-less pairing

girtsf, 2020-03-14:
The TTGO T-BEAM device does not come with a screen by default. Pairing currently displays a PIN on the device screen and then asks you to type it into the phone. We should set the right bluetooth flag when pairing to fall back to some simpler pairing mechanism if a screen does not exist.

spattinson, 2020-03-15:
One suggestion I thought of is to fall back to a fixed pin for pairing if there is no screen. I am testing without screens atm and use serial monitors for both nodes. I have ordered two screens though, cost about £3 each. Pairing is not a one time thing currently, it makes it a bit tricky without screen

geeksville, 2020-03-24:
good point, we can easily set a bit to use fixed pin if the device doesn't have a screen.  I'll try and do that today.

geeksville, 2020-03-25:
alas, fixed pin isn't supported yet in the esp arduino wrapper.  Adding button press based stuff would work but that is a bit different flow.  So for now I'm going to put to after 1.0 (unless someone else wants to do this).  We have enough work in the queue to support the current devices.

Professr, 2020-06-25:
Is the PIN exposed via the serial debug output?  If not, that might be a good addition in the interim until we can get proper support built

aHVzY2g, 2020-06-25:
Yes the PIN is exposed via Serial Debug.

mc-hamster, 2021-03-27:
Closed

This has been done. When prompted to pair, double click on the device button and the password will be set to 123456

closed 2021-03-27 by mc-hamster