meshtastic/firmware #27: make an installer script, that completely erases device flash

geeksville, 2020-03-14:
per https://meshtastic.discourse.group/t/erasing-flash-1st-may-be-important-433mhz-t-beam-working-for-me-now/29

Possibly explains bluetooth issues some have had

geeksville, 2020-03-14:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/erasing-flash-1st-may-be-important-433mhz-t-beam-working-for-me-now/29/2

spattinson, 2020-03-15:
The steps I made to get there were:

Build and  flash from platformio, do some testing, got a bt connection, then realise I am using US RF setting due to misunderstanding of how its set.
Rebuild with EU433 and flash.

Have issues with pairing, so clear data on the android app and delete android pairing.
Try to pair again, still having issues, pairing appears to succeed but the message in the android app about "You have not paired...." does not go away.

delete repo in my computer and pull down the latest, erase flash, flash, clear android app data, delete android pairings, so I know I am starting totally from fresh again, and test, it works. 

After a while I get "No connection" after the BT has gone to sleep, which I think you alluded to previously, and are addressing in a update to the Android app.

It may be the handling of flash data partition and stale pairings causing the issue?  It  could also be to do with what they've been flashed with at the factory. In the case of TT-Go T-Beam they are factory flashed with SoftRF firmware, the aviation collision avoidance tracker and flight recorder, which makes use of Bluetooth. It is my impression that users of that project are not all technically savvy to reflash their own units, they buy and use straight away. I sort of assumed that when you flashed that the partition map gets reinitialized by default, and if you don't specify you just get a default one. On the Arduino IDE one is selected whatever you do, whether esptool doesn't do this part sometimes if the existing partition map appears to match IDK?

commit f4d368e1 (2020-03-24): fix #27 - add a device-install.sh script to the release
closed 2020-03-24 by geeksville