meshtastic/firmware #38: Need documentation to add additional boards

SensorsIot, 2020-03-17:
Interesting project. I thought I could use it for my next video. I have many LoRa boards. Unfortunately not the ones you support. For example, I have a few of those: https://www.aliexpress.com/item/4000438284523.html or those: https://www.aliexpress.com/item/32825749403.html . 
When I upload the .bin file according to your instruction, the Heltec only reboots. If I upload your platformIO project to a T-beam without a display, at least I see that it runs in Serial.

I would like to get a description of the different pins to change and maybe to switch GPS off if not available. Then maybe I could add the boards I own.

geeksville, 2020-03-17:
re: your heltec board rebooting.
I think this is due to an error in the current instructions, can you test this theory by trying the following three commands and report back?
https://github.com/meshtastic/Meshtastic-esp32/issues/17#issuecomment-599734679

By my quick reading of the https://www.aliexpress.com/item/32825749403.html  link the key GPIOs are all the same as the HELTEC.  So I suspect the default flash layout (the other partions besides the app partition) were laid out differently by that manufacturer, so just doing those three steps I think should make it work.

ignore the following portion of this comment for now: I stopped editing it when I saw tha the TTGO board should work with the TTGO 0.1.6 or later image. Its pinouts seem the same as the TBEAM except it doesn't have a GPS or power management chip (and the tbeam load checks for those parts dynamically and just warns if missing).

re: porting to new board
Good idea.  How about for now (since I'm currently knee deep in finishing an android release) I put the tips here and later one of us merges it into a file in docs/software/porting.md.

To use your https://www.aliexpress.com/item/32825749403.html as an example... and looking at https://ae01.alicdn.com/kf/HTB18J0tSVXXXXb9XFXXq6xXFXXXW.jpg

Step 1:

I think only configuration.h needs t change.

```
#elif defined(HELTEC_LORA32)
// This string must exactly match the case used in release file names or the android updater won't work
#define HW_VENDOR "HELTEC"

#ifndef USE_JTAG // gpio15 is TDO for JTAG, so no I2C on this board while doing jtag
#define I2C_SDA 4
#define I2C_SCL 15
#endif

#define RESET_OLED 16

#define VEXT_ENABLE 21 // active low, powers the oled display and the lora antenna boost
#define LED_PIN 25
#define BUTTON_PIN 0

#ifndef USE_JTAG
#define RESET_GPIO 14
#endif
#define DIO0_GPIO 26
#define DIO1_GPIO 35
#define DIO2_GPIO 34
#endif
```

geeksville, 2020-03-17:
btw - sorry I can't help too much more right now, but I really want to get the current androdi changes out as a new release (today/tomorrow).  And everything is now going slower because of being in Corona lockdown mode in my town...

geeksville, 2020-03-24:
@SensorsIot Just checking in did that work out for you (alas, I was totally clobbered the last sixish days with either flu or covid and I'm finally feeling human)

SensorsIot, 2020-03-24:
Glad to read you are better. No, I had to give up for the moment. Because I have a schedule of my videos I always have to assess if I am able to deliver ;-)
I am still interested, but I might need some help from you.
I would like to show a scenario with two (better 3) devices exchanging messages. 
Would you be willing to skype with me? My name is spiessa.

geeksville, 2020-03-25:
Happily.  Can we work out a time by email?  I'm kevinh@geeksville.com and I'm available virtually any time after Thursday.  Currently crunching on the next android release.

closed 2020-04-03 by geeksville