meshtastic/firmware #49: Make debug info screen show real data

girtsf, 2020-03-19:
- [ ] Show battery level & charging status

girtsf, 2020-03-19:
@geeksville - what else?

geeksville, 2020-03-19:
oh yeah - battery level would be great (depending on what the axp192 can give you - it might be only voltage, which is certainly better than nothing - there are good tables on the web that will let you map lipo voltage to % full (poorly))

Also showing the GPS lock/no-lock would be great.

spattinson, 2020-03-19:
For battery level axp.getBattPercentage() doesn't work. I dont think its supported on axp192, only axp202 so have to calculate it from voltage eg 4200mv = 100% etc. Currently battery percent on the display is 0% = not charging and 1% = charging/on AC

geeksville, 2020-03-24:
@spattinson yep - thanks - I was guessing that but I'm glad you checked. If either you or girts want to do this google for "lipo battery voltage to percentage full" and there are some good tables that provide a crummy (but better than the current approach) approximation.

Really someday we'd need to do coulomb counting by polling the axp current/voltage registers during use and charging and integrate an approximation.

spattinson, 2020-03-25:
Coulomb counting looks complicated to me. As we are dealing with a low current draw we can determine battery percentage reasonably accurately from the voltage alone. While the battery is charging the apparent voltage is boosted a bit depending on the charge current, could subtract a few percent, then when the cell is at 4.2V look at the current dropping and wait for it to stop charging before displaying 100%.  By default the axp192 considers the battery empty and cuts it off to prevent over discharge at 2.8V, it would be better for the battery if that was a bit higher perhaps 3.0v. This is set at  Reg 31H according to datasheet. This csv is a discharge test I did on my vape on a fairly new LG HG2 18650 3000mAh, its probably close to typical for good quality cells.
[lg_hg2_csv.zip](https://github.com/meshtastic/Meshtastic-esp32/files/4382436/lg_hg2_csv.zip)

geeksville, 2020-03-25:
@spattinson that is super useful advice and I agree.  Btw, no pressure but it sounds like you have more experience with this power management chip than either I or girts.  If you ever feel like auditing the (really bad, based on ttgo example code that doesn't even match the datasheet) axp192 code in main.c that would be super appreciated (but no pressure).  See also bug #48.

spattinson, 2020-03-25:
I will try to reproduce #48 with the original sample sketch and see if that works and have a fiddle with it.

commit 54cd082b (2020-03-27): fix #49: make debug screen show real data

* Break out debug screen to a separate class and make it thread-safe.
* Break out power state to a separate class.
* Show battery voltage, charging & USB status on debug screen.
* Show GPS lock / no lock
* Fix an off-by-one that I introduced earlier in `drawRows`.
commit f8857ad4 (2020-03-27): Merge pull request #63 from girtsf/debug-screen-b49

fix #49: make debug screen show real data
closed 2020-03-27 by geeksville