meshtastic/firmware #50: Oled is upside down

claesg, 2020-03-19:
Flashed one of my T-Beam's with 0.1.8
The screen is upside down compared to the other T-Beam which is running 0.1.7
[image]
The right one in the picture is running 0.1.8
The T-Beam is a V1.0.

girtsf, 2020-03-19:
@geeksville - on the T-BEAM that I have, the new orientation looks better (due to the two-color display). But perhaps for majority of people the old orientation makes more sense? Should we revert to the old behavior and have an opt-in define to flip it?

geeksville, 2020-03-19:
In fact the one you have girts was a test device I built up - all of the
standard tbeams are monocolor.  So probably just good to turn that ifdef
off for everybody.  The next time I see you we can swap tbeams and I'll
take that screen off and put on the standard one.

(Sent from a phone - please ignore typos)

PR #51 (2020-03-24): screen.cpp: flip the display 180
Fixes #50.
closed 2020-03-24 by geeksville