meshtastic/firmware #53: rapidly repeating message: canSleep,

spattinson, 2020-03-21:
version 0.1.8 -It was happening on previous version too, this time about ten minutes after flashing. I dont know if its always the same node. I will check this time it was my node ending 44.
```
canSleep, mode=3, isRx=0, txEmpty=0, txGood=4
canSleep, mode=3, isRx=0, txEmpty=0, txGood=4
canSleep, mode=3, isRx=0, txEmpty=0, txGood=4
Trigger powerFSM 9
Transition powerFSM transition=Contact from phone, from=DARK to=DARK
pbread from ea9f3f82-8dc4-4733-9452-1f6da28892a2 returns 30 bytes
canSleep, mode=3, isRx=0, txEmpty=0, txGood=4
canSleep, mode=3, isRx=0, txEmpty=0, txGood=4
canSleep, mode=3, isRx=0, txEmpty=0, txGood=4
canSleep, mode=3, isRx=0, txEmpty=0, txGood=4
canSleep, mode=3, isRx=0, txEmpty=0, txGood=4
canSleep, mode=3, isRx=0, txEmpty=0, txGood=4
``
previously I didn't see the message in the middle, its usually just the other lines scrolling very quickly. Cannot get screen to turn on or anything happening by pressing the button. Blue led is off, have to press RST button then it works again. I am tending to leave my units on all they time, this problem usually arise some minutes after boot. I will see if I can capture the preceeding message by running the terminal through tee, or make it log to file

geeksville, 2020-03-24:
oh that is super interesting.  I'll write more on this bug later tonight (just got back from sickness)

geeksville, 2020-03-24:
ok, checked the code, this means we have a bug (likely a race condition) where the lora radio is staying stuck in transmit mode (rather than returning to rxmode after sending).  I'm now looking to try to find the cause and something to check for it and report the bug earlier.

geeksville, 2020-03-24:
@spattinson I'm releasing a build later today 0.1.9 that I think probably fixes this (though I'm having a hard time reproducing it).  I fixed one possible root cause, but also added a watchdog to see if this bug occurs, if it does it 'fixes' it and reports back to the mothership (eventually).  

Can you check with 0.1.9 if it prints "ERROR! Bug! Tx packet took too long to send, forcing radio into rx mode"? that will tell me if the root bug still exists?  (though it will also apply a workaround to make the bug not fatal)

spattinson, 2020-03-25:
Yes sure, here are two logs, I think only 2c.txt exhibits the issue, 44.txt is its partner node, there is also a backtrace in there, I don't know if that is related?
[logs.zip](https://github.com/meshtastic/Meshtastic-esp32/files/4382282/logs.zip)

geeksville, 2020-03-25:
yeah - that was super helpful.  Two interesting things:

I see rxBad going up, which should almost never happen (I have very few packets with bad crc received on my test bench) and twice it screwed up and tried to send when we were already sending.  I have a bug that needs fixing and will work on this later today.

for my reference the relevant bits are:

```

fix type 3
fix type 3
fix type 3
new gps pos lat=51.480528, lon=-0.616718, alt=52
got gps notify
Sending position to mesh
Update DB node 0x2c for variant 1, rx_time=1585144143
Providing time to mesh 1585144143
immedate send on mesh (txGood=2,rxGood=0,rxBad=1)
fix type 3
new gps pos lat=51.480466, lon=-0.616749, alt=58
got gps notify
Update DB node 0x2c for variant 1, rx_time=1585144173
ERROR! Bug! Tx packet took too long to send, forcing radio into rx modefix type 3
new gps pos lat=51.480512, lon=-0.616759, alt=61
got gps notify
Update DB node 0x2c for variant 1, rx_time=1585144203
Transition powerFSM transition=Bluetooth timeout, from=DARK to=LS
lsEnter begin, ls_secs=3600
canSleep, mode=4, isRx=0, txEmpty=1, txGood=2
lsEnter end
lsIdle begin ls_secs=3600
wakeCause 7
Trigger powerFSM 2
Transition powerFSM transition=Wake timer, from=LS to=DARK
Looking for GPS lock
fix type 3
new gps pos lat=51.480459, lon=-0.616718, alt=60
got gps notify
Update DB node 0x2c for variant 1, rx_time=1585144223
fix type 3
fix type 3
fix type 3
fix type 3
fix type 3
fix type 3
fix type 3
fix type 3
fix type 3
Transition powerFSM transition=Bluetooth timeout, from=DARK to=LS
lsEnter begin, ls_secs=3600
canSleep, mode=4, isRx=0, txEmpty=1, txGood=2
lsEnter end
lsIdle begin ls_secs=3600
wakeCause 7
Trigger powerFSM 2
Transition powerFSM transition=Wake timer, from=LS to=DARK
Looking for GPS lock
fix type 3
fix type 3
fix type 3
fix type 3
new gps pos lat=51.480459, lon=-0.616718, alt=60
got gps notify
Update DB node 0x2c for variant 1, rx_time=1585144373
fix type 3
new gps pos lat=51.480459, lon=-0.616718, alt=60
got gps notify
Update DB node 0x2c for variant 1, rx_time=1585144403
fix type 3
fix type 3
new gps pos lat=51.480459, lon=-0.616718, alt=60
[... 113 lines cut ...]
fix type 3
new gps pos lat=51.480525, lon=-0.616834, alt=56
got gps notify
Update DB node 0x2c for variant 1, rx_time=1585144993
fix type 3
fix type 3
fix type 3
fix type 3
new gps pos lat=51.480525, lon=-0.616834, alt=56
got gps notify
Sending position to mesh
Update DB node 0x2c for variant 1, rx_time=1585145053
Providing time to mesh 1585145053
immedate send on mesh (txGood=2,rxGood=0,rxBad=2)
assertion "!sendingPacket" failed: file "src/CustomRF95.cpp", line 155, function: void CustomRF95::startSend(MeshPacket*)
abort() was called at PC 0x4010362f on core 1

Backtrace: 0x40094cb4:0x3ffd8f80 0x40094ee5:0x3ffd8fa0 0x4010362f:0x3ffd8fc0 0x400d2686:0x3ffd8ff0 0x400d2789:0x3ffd9010 0x400d3d51:0x3ffd9040 0x400d3ed1:0x3ffd9070 0x400d4469:0x3ffd9090 0x400d4492:0x3ffd90b0 0x400d2ba9:0x3ffd90d0 0x400d4d63:0x3ffd9130 0x400d29b9:0x3ffd9160 0x400d5993:0x3ffd9180 0x400e78ad:0x3ffd91b0 0x400913c9:0x3ffd91d0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
```

geeksville, 2020-03-25:
actually - looking more, only screwed up once.  My little emergency txwatchdog was forcing the radio into rxmode correctly, but not clearing the flag that said we were busy sending.

geeksville, 2020-03-25:
also, I see you were running a local build.  I just noticed that platform IO doesn't pull down the latest git dependencies unless you do "platformio update", so you might not have had my fixes in the RadioHead library.  But I'll still think about this problem later today.

spattinson, 2020-03-26:
I have updated radiohead, it was behind:
Updating RadioHead                       @ 57ac37e        [0b279d3]
I have made some changes to my local copy for #48 and updated that issue. I have those changes and updated RadioHead so I will leave it logging again. I am just leaving them on all the time and periodically waking with button and send a msg from app

geeksville, 2020-04-05:
@spattinson btw: I just reproduced the original problem on one of my boards.  Your your 0x2c board that showed the problem, did that board have a battery installed?  On my board which can show the problem - it occurs 100% of the time if I don't have a battery installed, 0% of the time if I do have one installed.  I'm beginning to suspect there is something marginal in the power rail that powers the  LORA radio and with no battery, if the USB power isn't up-to-snuff the radio never sets TX complete.

spattinson, 2020-04-07:
All my testing is being done with batteries, I basically leave them on all the time now, and plugged in. I have not witnessed the problem recently, but I have not been watching closely either, I am mostly tee'ing output to logs and there are no reboots in them.

geeksville, 2020-04-08:
ok sounds good!  Sounds like bug probably squashed.  Thanks!

closed 2020-04-08 by geeksville