meshtastic/firmware #59: PowerFSM onEnter calls sendNetworkPing(broadcast), that is sub optimal

geeksville, 2020-03-25:
When the user presses the button we currently ask all nodes for an update.  Instead of calling sendNetworkPing(BROADCAST) we should be calling it with the ID of the node that the user is currently displaying in the display.

Fix this by adding a getDisplayedNode() function to Screen.  That function should return null if we are not looking at a node, otherwise it should return the node we are watching.  in PowerFSM onEnter() if that node is null do nothing, otherwise send the ping only to that node.

BeardyWalrus, 2020-04-10:
I've been taking a quick look into this. Is it best to read the node displayed on the screen, or would it work if the nodedb kept a record of the node that is currently displayed or otherwise to be notified on wake. Maybe even going so far as to be able to set specific nodes that should be notified on wake via bluetooth api.

geeksville, 2020-04-11:
@BeardyWalrus I think for now either of the first two options would be fine.  i.e. either have screen set a "mostRecentViewedNode" property of NodeDB.  And when we press the button ping that node.

Or might might be more natural (because buttons and screens have a close relationship), to just ask the screen "what is the currently shown device".  I tend to think this option is better.

Thanks for looking at this btw.

commit d2de04d5 (2020-05-22): Fix #59 no need for broadcasts when showing new node pane
closed 2020-05-22 by geeksville