meshtastic/firmware #60: Add support for a 'I need emergency help' button.

geeksville, 2020-03-25:
This is a short user story, so that we don't forget this idea and do it sometime post beta.  It would need a fair amount of user testing and feedback and proceed in stages.

Stage 1: User falls and hurts their ankle, they long press on the button.  The device prompts "release and press three more times to call for help." After the user presses three times the device enters emergency mode.

While in emergency mode all the normal features work the same, but a location packet with the "I need help" bit is also broadcast to the entire mesh every 5 minutes.

When another node receives this packet the screen lights up (showing direction and distance and help) and led flashes.  The associated phone (if connected) will chime with a notification.  Presumably this will prompt the users friends who are on this same channel to ask what's up.  One of those users takes responsiblity for messaging the injured person and calling 911 as needed/possible.

After discussing stage 1 with other users/devs and implementing stage one we get feedback on utility (by this point we will probably have a couple of hundred users of the beta - and real data on how these devices are working for users on hikes etc).

If Stage 1 goes well, Stage 2: Add noonlight support to the app.  In this case the flow would be as above except any of the members who have cell coverage would be offered the option to "Report this problem to 911", then the user would be asked to double confirm.  Then we'd use the API to put in a call giving the gps coords _of the person who had the accident_ but the contact information _of the person who is acting as their agent for talking to 911_.  The noonlight service would then call/text the person who reported the accident and get more info, tripple confirm this is a real emergency etc... and contact first responders.

This user story was prompted because I was contacted by a product manager for [Noonlight](https://www.noonlight.com/).  After a useful email conversation between the two of us I decided to capture this as an issue, so that we can work for this after the beta release.

geeksville, 2020-03-25:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/user-story-added-for-an-i-need-help-mode/65/1

geeksville, 2020-08-10:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/public-emergency-chat-room-not-encrypted/713/14

android606, 2021-01-29:
Perhaps pressing the "I need help" button **once** means "help, I'm lost, come find me", but pressing it **three times** or something similar means "help, my life is in danger". The first case sends an alert to everyone in the mesh, with current/last known location. The second case sends an alert, or possibly ongoing periodic alerts, AND prompts the rest of the mesh users to contact 911 / emergency responders. That way, the "I need help" button could be used by a kid/person that accidentally took the wrong trail fork to notify the others, WITHOUT accidentally sending in the Search & Rescue helicopter.

mc-hamster, 2021-02-23:
Just thinking about it ... this is almost integrated into the ExternalNotificationPlugin as is.

We can take advantage of the bell character as part of the current implementation, so when a device gets the bell character, that means "SOS!"

On the device hardware, it would mean adding support for an event to send the emergency button and then adding support to the mobile device to respond to the alert character.

mc-hamster, 2021-02-23:
In fact, we can just work with the existing button with button.attachMultiClick and if we get five quick clicks, that means we want to send an alert. That would mean no additional hardware.

garthvh, 2021-12-29:
Support for bell character https://github.com/meshtastic/Meshtastic-device/issues/1035

mc-hamster, 2022-01-17:
Closing this as the device side is done.

Triggering on the ASCII BELL will cause an alarm if used with the external notification plugin.

ASCII BELL can be sent by a client as a tap back and/or in the message itself.

More support needs to be built out in the clients.

At present, there's no way to send the ASCII BELL from the device directly, but that can be a feature request of the ext notification plugin if it's needed.

closed 2022-01-17 by mc-hamster