meshtastic/firmware #62: Feature idea: Compose messages without a paired phone

myself248, 2020-03-26:
Say your phone battery (1-2 days) is dead, but your Meshtastic battery (7-8 days) is still going strong. It would be nice to have some minimal message-composition functionality, even if it's slow and awkward.

Several ideas come to mind: 

Alphabet on screen, like "high-score-entry" on an arcade game, that you can navigate with a few buttons. Slowest, but probably simplest to implement. 

Morse-code with a single button, for folks who know the code. There's plenty of existing code that does this already, but a lot of it sucks, and I don't know how much work it would be to find a good one and integrate it. 

A menu of canned phrases with optional blanks to fill in, or words to flip like "did/didn't". There exist various brevity codes that may provide inspiration, or make up our own. I'm picturing categories like "location and people", "status/health", "weather/conditions", "need/find", "radio technical". So you navigate into the "location and people" category and there are phrases like "I've [not] reached my destination", "Please [don't] come [to my lat/lon]", "What is your ETA to [where you're going|here]?", "I'm here with ___", "Where would I find ___?", and so on. Pick one and you can then modify it before sending.

jeksys, 2020-03-26:
Garmin InReach mini has proposed feature. Small onscreen keyboard and predefined custom messages

geeksville, 2020-03-27:
yah - i'm a big fan of the inreach style of UI.  let the phone app edit and supply a set of customized messages.  Then on the device let the user scroll and click to send a canned message (from that list)

Professr, 2020-06-25:
What about bluetooth HID?  There are some perfect pocket bluetooth keyboards that would probably work great for text entry without a phone

android606, 2021-01-29:
The InReach method is awesome for extremely low-data **yes, I made it out** / **no, I died** type messages. LoRa doesn't need to be nearly so limited, though. This request seems like two different but related features: 
1) Allow user to pre-program a half-dozen messages from the app that they can later send directly from the device.
2) Allow user to add an optional input device (I2C keyboard? Bluetooth keyboard?) and use it to type and send messages directly from the device.