meshtastic/firmware #64: make a supported devices page with pictures

geeksville, 2020-03-27:
and instructions on where to buy, how to solder in screen if it didn't come with one.

move some of this contents from README.

per comment from @girtsf

geeksville, 2020-03-28:
@spattinson, I just received this great spreadsheet from @sensorsiot:

I'll give anyone (for now) who asks (just click and request access) permission to edit it.

https://docs.google.com/spreadsheets/d/1Qw7RIcm8IRcoS0vGVAa7YqeavwGAmRtPEgvcR7ABaIE/edit?usp=sharing

It includes some new devices I'm working on adding support for.  I'll also add our existing devices.  It could be useful for your effort - I wonder if for the "table of devices" page if we even need to make markdown for the table portion.  Would users like it just as much as a link to this google spreadsheet?

spattinson, 2020-03-28:
I can add a table with the pins and features . I already made reference to those boards, but said that user will have to add them to configuration.h and make an entry in platformio.ini. I think thats all esp32 lora boards I know of covered. It looks like the TTGO LORA 2.1  has resistor divider and analog voltage monitoring. If anyone uses that they will probably ask for that functionality to be added, obviously after the ax192 battery meter, either way both will be voltages so they can share the conversion to percent. If anyone has the older boards they can solder on the resistors to get a battery gauge. And a button if they want, how do boards without buttons wake from light sleep after BT is turned off?

spattinson, 2020-03-29:
@geeksville This is what I've done so far https://github.com/spattinson/Meshtastic-esp32/wiki/Hardware-Information let me know any additions/changes?
I could pull all your other documents together and make some sections for wiki then all the info is in one place?

lburrowes, 2020-03-29:
[image]

[image]

android606, 2020-03-30:
@spattinson I wish I had your Hardware-Information document two days ago when I was combing through the code to try to figure out how to solder a display onto my T-Beam! I was really not sure if I was doing the right thing or not when I was bending the header pins...

[image]

spattinson, 2020-03-30:
If you 3d print a case, aligning it with the hole for screen needs to Gnd 3.3V to be bent not the I2C pins so it shifts over to the right

geeksville, 2020-03-30:
How @spattinson - that is amazing and a far better job than I would have
done!

geeksville, 2020-03-30:
also, I've just changed the permissions to allow public editing of the meshtastic-esp32 wiki.  Are you okay with me merging your changes into that wiki now?

android606, 2020-03-30:
Feel free to use that image above if you want, since it does show only the GND and 3v3 pins bent.

spattinson, 2020-03-30:
@android606 Thanks. Thats better, bending GND and 3.3V aligns the screen with how TT-Go do it, and lines up with the holes on the 3d printer models.
@geeksville I've put all the images in a subfolder inside wiki on my fork so I dont need to add them to your top level images folder and to keep them separate. I can't seem to git push it to here though, get 403, I would just make a folder through the web interface and do it that way but there doesn't seem to be a way to create folders in the wiki. I can create pages in the wiki through the web interface fine.

geeksville, 2020-03-31:
btw: these screens have the pins in the right place to not need bending
(they also sell singles):
https://www.amazon.com/gp/product/B0837DLWVH/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

geeksville, 2020-03-31:
ok merged!  Thank you so much @spattinson.  great work!

btw: if you want to edit these via git in the future (rather than the github webpage based editor) just do a git clone from https://github.com/meshtastic/Meshtastic-esp32.wiki.git

closed 2020-03-31 by geeksville

Dafeman, 2020-04-07:
By the looks of it, if you have a board like [this](https://pasteboard.co/J2GrRIA.jpg) R3 and R4 can be reversed to route pins 1 and 2 in reverse, allowing the pins to line up correctly.

spattinson, 2020-04-07:
Good spot!  I guess they are zero ohm so rather than actually move them just take off and bridge with solder may be easier
[image]
I've added to wiki, thanks!

Dafeman, 2020-04-07:
Yea definitely 0ohm. Solder bridge is a good option! Thanks for adding to the Wiki.

geeksville, 2020-04-08:
ooh! that is a good find @Dafeman