meshtastic/firmware #66: Voice over LORA.  ...sort of.

DeadlyDad, 2020-03-27:
It just struck me this morning (while pondering another project) that one way to transmit voice over LORA would be to use cellphone apps on either end to do speech to text on one end, send the text, then do text to speech on the other.  (Might it be possible for individuals to set up phoneme databases so that the transmitted voice would be theirs?)  One further advantage would be a transcript of the conversation.  (...and a lot of laughter from autocorrect hijinx.  :grin: )

Does that make sense?

jeksys, 2020-03-27:
What's the max/avg bandwidth for LORA?

geeksville, 2020-03-27:
Alas I think this is not a good for for Lora.  Because when configured for
long range, the number of chirps per bit becomes quite long.

I.e. sending a 100 byte packet takes a couple of seconds.

And there are maximum duty cycle limits for transmitters for most
regulators.  For our 'occasional' gps and text messages no problem, but if
you squirt many packets quickly you could have problems.

(Sent from a phone - please ignore typos)

DeadlyDad, 2020-03-27:
> What's the ~max/avg~ realistic bandwidth for LORA?

I'm not sure why you are asking me that, but there isn't a simple answer: https://medium.com/home-wireless/testing-lora-radios-with-the-limesdr-mini-part-2-37fa481217ff

If you want to know more, I invite you to do some research on your own.  (I'm **not** a LORA expert.)

geeksville, 2020-03-27:
Oh I see.  You mean speech to text and text to speech then send
characters.  Yes, that would be possible.   Either with a separate app that
uses service we have running on the phone or by sending in a PR for the app
itself.

(Sent from a phone - please ignore typos)

DeadlyDad, 2020-03-27:
Thanks for the reply.  I suppose 400bps or less would be workable if communications were kept short and folks understood the limitations.   (...and if you could figure out how to get a fairly reliable 700bps, you could have actual VOIP.  ...ish: https://hackaday.com/2017/01/20/voice-at-700-bits-per-second/

jeksys, 2020-03-27:
@DeadlyDad thank you detailed answer :)

100 byte packet per couple of seconds is not a lot. 

Most of speech recognition frameworks require network connection. 
For iOS:
"On-device speech recognition is available for some languages, but the framework also relies on Apple’s servers for speech recognition. Always assume that performing speech recognition requires a network connection."
https://developer.apple.com/documentation/speech

There are frameworks which doesn't require connection. OpenEars is one of them, though I'm not sure if it's good enough.

geeksville, 2020-03-28:
agree.  I'm going to move this into the "good idea which we would happily accept PRs for, but won't have time to ponder ourselves" queue.

DeadlyDad, 2020-03-28:
(While _some_ may require an internet connection, I was using voice dictation software on a Pentium running Windows 95 before the internet was even *available* where I lived, and there are a number of projects that don't require a network connection.  Here, for example, are five of them: https://fosspost.org/alternative-software/developers/open-source-speech-recognition-speech-to-text )