meshtastic/firmware #68: Core Panic on Firmware .2

lburrowes, 2020-03-29:
TTGO LORA32 V1

Turning on screen
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c46c  PS      : 0x00060330  A0      : 0x800d8b40  A1      : 0x3ffd90b0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000400  A5      : 0x00000000
A6      : 0x0000003a  A7      : 0x00000040  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000050  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x0000003f

Backtrace: 0x4000c46c:0x3ffd90b0 0x400d8b3d:0x3ffd90c0 0x400d93f8:0x3ffd90e0 0x400d949d:0x3ffd9100 0x400d6613:0x3ffd9130 0x400d4a33:0x3ffd9160 0x400d55a7:0x3ffd9190 0x400e5945:0x3ffd91c0 0x4008ee31:0x3ffd91e0

girtsf, 2020-03-29:
This is what the backtrace decodes to:
```
$ bin/exception_decoder.py -e ~/Downloads/firmware-0.2.0/firmware-ttgo-lora32-v1-US-0.2.0.elf st
stack:
0x400d8b3d: OLEDDisplay::clear() at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/ttgo-lora32-v1/ESP8266_SSD1306/src/OLEDDisplay.cpp:813
0x400d93f8: OLEDDisplayUi::tick() at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/ttgo-lora32-v1/ESP8266_SSD1306/src/OLEDDisplayUi.cpp:219 (discriminator 4)
0x400d949d: OLEDDisplayUi::update() at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/ttgo-lora32-v1/ESP8266_SSD1306/src/OLEDDisplayUi.cpp:219 (discriminator 4)
0x400d6613: meshtastic::Screen::doTask() at /home/kevinh/development/meshtastic/meshtastic-esp32/src/screen.cpp:540
0x400d4a33: PeriodicTask::loop() at /home/kevinh/development/meshtastic/meshtastic-esp32/src/PeriodicTask.cpp:18
0x400d55a7: loop() at /home/kevinh/development/meshtastic/meshtastic-esp32/src/main.cpp:428
0x400e5945: loopTask(void*) at /home/kevinh/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
0x4008ee31: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

geeksville, 2020-03-29:
Thanks @girtsf and @lburrowes.  That report let me found the bug.  Checking in a definite fix.

Btw @lburrowes, can you physically look at your LORA board and tell me which picture matches? https://docs.google.com/spreadsheets/d/1Qw7RIcm8IRcoS0vGVAa7YqeavwGAmRtPEgvcR7ABaIE/edit#gid=643895886

closed 2020-03-29 by geeksville

lburrowes, 2020-03-29:
here is the model I have on hand.

[https://primalcortex.files.wordpress.com/2017/11/ttgofront.jpg?w=300&h=161](url)
[https://primalcortex.files.wordpress.com/2017/11/ttgoback.jpg?w=300&h=142](url)

geeksville, 2020-03-30:
@lburrowes Interesting!  I think you've found _yet another variant_ of these boards.  From looking at your stack trace I fixed a problem where we would crash if we didn't find a display at start (checked in but not released as a bin yet).  Would you mind trying the 0.2.0 build lora32-v2 build on your board?  The relevant change on that board is that it looks for the OLED using i2c on pins 4/15 instead of 21/22.

lburrowes, 2020-03-30:
I will load it up in a few minutes here and follow up.  I am in full testing mode...

commit 11d57e72 (2020-03-29): fix #68 (@girtsf, pls review - ps: no worries ;-) )

// We don't set useDisplay until setup() is called, because some boards have a declaration of this object but the device
// is never found when probing i2c and therefore we don't call setup and never want to do (invalid) accesses to this device.