meshtastic/firmware #73: Android App Causing Radio to Reboot

lburrowes, 2020-03-31:
After pairing, in the setting I see the radio.  When I select other menu items it just sends the radio into a boot loop.

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
booted, wake cause 0 (boot count 1), reset_reason=reset
I2C device found at address 0x3c
ssd1306 display found
done
Meshtastic swver=0.2.0, hwver=1.0-US
Turning on screen
Read RTC time as 29 (cur millis 89) valid=0
ERROR: No bidirectional GPS found, hoping that it still might work
Loading saved preferences
Warn: devicestate is old, discarding
NODENUM=0x4, dbsize=1
Starting meshradio init...
IRQ flag mask 0x0
Set radio: name=Default. config=3, ch=6, txpower=23
sending owner !246f28978300/Unknown 8300/?00
Update DB node 0x4 for variant 4, rx_time=0
old user !246f28978300/Unknown 8300/?00
updating changed=0 user !246f28978300/Unknown 8300/?00
immedate send on mesh (txGood=0,rxGood=0,rxBad=0)
Screen: Started...
showing standard frames
Transition powerFSM transition=boot timeout, from=BOOT to=ON
Setting bluetooth enable=1
Pre BT: 194964 heap size
Starting bluetooth
*** Mesh service:
 onAuthenticationComplete -> success size: 12
showing standard frames
Trigger powerFSM 9
Error: can't encode protobuf unterminated string
assertion "0" failed: file "src/mesh-pb-constants.cpp", line 17, function: size_t pb_encode_to_bytes(uint8_t*, size_t, const pb_msgdesc_t*, const void*)
abort() was called at PC 0x400fea63 on core 0

Backtrace: 0x4009271c:0x3ffe2070 0x4009294d:0x3ffe2090 0x400fea63:0x3ffe20b0 0x400d577e:0x3ffe20e0 0x400d314a:0x3ffe2120 0x400d31db:0x3ffe2170 0x400d31f2:0x3ffe2190 0x400dcda1:0x3ffe21b0 0x400e1135:0x3ffe2460 0x400df911:0x3ffe2480 0x400dfd0d:0x3ffe24d0 0x400df361:0x3ffe24f0 0x400ddcbd:0x3ffe2550 0x4011dca1:0x3ffe2570 0x40117d52:0x3ffe25b0 0x4008ee31:0x3ffe25e0

Rebooting...

geeksville, 2020-03-31:
investigating.  stack trace decoded as:

kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$
bin/exception_decoder.py -e
release/archive/firmware-ttgo-lora32-v2-US-0.2.0.elf ~/tmp/stack
stack:
0x4009271c: invoke_abort at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
0x4009294d: abort at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
0x400fea63: __assert_func at
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63
(discriminator 8)
0x400d577e: pb_encode_to_bytes(unsigned char*, unsigned int, pb_msgdesc_s
const*, void const*) at
/home/kevinh/development/meshtastic/meshtastic-esp32/src/mesh-pb-constants.cpp:17
0x400d314a: ProtobufCharacteristic::onRead(BLECharacteristic*) at
/home/kevinh/development/meshtastic/meshtastic-esp32/src/MeshBluetoothService.cpp:37
0x400d31db: MyNodeInfoCharacteristic::onRead(BLECharacteristic*) at
/home/kevinh/development/meshtastic/meshtastic-esp32/src/MeshBluetoothService.cpp:237
0x400d31f2: non-virtual thunk to
MyNodeInfoCharacteristic::onRead(BLECharacteristic*) at
/home/kevinh/development/meshtastic/meshtastic-esp32/src/MeshBluetoothService.cpp:240
0x400dcda1: BLECharacteristic::handleGATTServerEvent(esp_gatts_cb_event_t,
unsigned char, esp_ble_gatts_cb_param_t*) at
/home/kevinh/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLECharacteristic.cpp:702
0x400e1135:
BLECharacteristicMap::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned
char, esp_ble_gatts_cb_param_t*) at
/home/kevinh/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLECharacteristicMap.cpp:87
0x400df911: BLEService::handleGATTServerEvent(esp_gatts_cb_event_t,
unsigned char, esp_ble_gatts_cb_param_t*) at
/home/kevinh/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLEService.cpp:390
0x400dfd0d: BLEServiceMap::handleGATTServerEvent(esp_gatts_cb_event_t,
unsigned char, esp_ble_gatts_cb_param_t*) at
/home/kevinh/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLEServiceMap.cpp:135
0x400df361: BLEServer::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned
char, esp_ble_gatts_cb_param_t*) at
/home/kevinh/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLEServer.h:119
0x400ddcbd: BLEDevice::gattServerEventHandler(esp_gatts_cb_event_t,
unsigned char, esp_ble_gatts_cb_param_t*) at
/home/kevinh/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLEDevice.cpp:570
0x4011dca1: btc_gatts_cb_to_app at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gatts.c:54
  \-> inlined by: btc_gatts_cb_handler at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gatts.c:762
0x40117d52: btc_task at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:110
0x4008ee31: vPortTaskWrapper at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355
(discriminator 1)

geeksville, 2020-03-31:
oh - I see.  "ttgo-lora32-v2" is longer than what I thought a max device
name would be. Thanks for the report - fixing now.

commit 17de6f95 (2020-03-31): fix #73: allow hw-model to be longer (16 bytes including terminator)
closed 2020-03-31 by geeksville