meshtastic/firmware #88: Good ideas for docs or videos new developers could contribute ;-)

geeksville, 2020-04-15:
doc/video ideas:

It would be awesome if someone wants to make these happen!

* how to pair with phone, how to use GUI on phone, how to send/receive a text message
* how to use gui on device, how to navigate to someone else
* how to understand the serial log output on the device, what messages mean phone is talking etc...

(Based on discussion with @sensorsiot)

ohcdh, 2020-04-15:
I will happily do these.  I am still waiting on my boards to arrive, if it would be helpful or if someone else wants to do in the meantime, please go ahead.  Once my boards arrive I will be making a few videos/docs on getting cases, assembling, flashing etc.

geeksville, 2020-04-15:
That would be awesome!  Someone was just asking for this.

(Sent from a phone - please ignore typos)

b3hr3ns, 2020-04-20:
I've subscribed to this!  I would love to be able to contribute more so than just having 4 radios, 4 phones and uploading logs when something doesn't go right.  It doesn't feel right just pointing out errors and not doing anything about them. 

Currently, to make it easier getting logs, I have all 4 radios connected to an asus tinkerboard all contained nicely in a small plastic project box with one power supply. From the tinkerboard (similar to a raspberry pi) I can get USB serial output over SSH, then copy-paste text files here. 

Can't wait to see what docs you come up with!  From what I've already seen from the code, I just need to read the comments and really, really dig in.

geeksville, 2020-04-21:
That would be awesome.  We'd love to have any help you can offer.  Keep
rocking it.  Cool setup.

geeksville, 2020-04-29:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/quick-start-guide/157/2

hjenemark, 2020-07-24:
Hi.
I'm thinking about looking into the 3 buttons of the TBeam board. 
Are those described anywhere?

And what is the format of the documentation?
ReadMe files on github?

geeksville, 2020-07-24:
@hjenemark The three buttons (from left to right) are "power on", "next", and "reset".  Only one of which is visible by software "next".  So all UI is through that button.  A _very_ long press on power-on will force the power off, but you should never need to press that.  If you want to add this to the wiki that would be great: https://github.com/meshtastic/Meshtastic-device/wiki.  You can click to edit "hardware information"

yhaenggi, 2020-07-24:
If you hold the "next" button, you can change the brightness on tbeams (didn't test on others).

closed 2021-12-12 by sachaw