meshtastic/firmware #93: ttgo-Lora32-V2.1 1.6Version has no firmware

zsp1123, 2020-04-17:
https://github.com/Xinyuan-LilyGo/TTGO-LoRa-Series
https://www.aliexpress.com/item/32915894264.html?spm=2114.12010612.8148356.6.2f4c3615hn7e2j

and i can't find the bt pairing code ,0000 and 1234 are both wrong.

geeksville, 2020-04-17:
I think for that board you want firmware-ttgo-lora32-v2-* from our release binaries.  Some of the TTGO lora boards are mislabeled and you actually want firmware-ttgo-lora32-v1-*.  You'll know you have the right board because you'll see the meshtastic logo on the screen.

When pairing the pairing code will be displayed on the OLED screen (or for devices that don't have a screen, on the serial port.

zsp1123, 2020-04-17:
thanks for you help, i flash firmware-ttgo-lora32-v1-EU433 , the OLED and bt can work

closed 2020-04-17 by zsp1123

geeksville, 2020-04-17:
btw - if you have any future questions there is a thread on our forum about the TTGO-LORA32 devices with some useful info.