meshtastic/firmware #95: Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited)

spattinson, 2020-04-17:
I've got this on both nodes, I did a git pull yesterday, and built myself - I notice there are a few commits since the last release zip, its in MeshService.cpp:204 which is handleFromRadio();
Here is stack and decode from both, they are very similar, it could also be some external library/framework though,:
```
scott@n550jk:~/pio/Meshtastic-esp32$ bin/exception_decoder.py -p ESP32 -e .pio/build/tbeam/firmware.elf /tmp/crash1
stack:
0x400d39d5: MeshService::loop() at /home/scott/pio/Meshtastic-esp32/src/MeshService.cpp:204
0x400d5ff3: loop() at /home/scott/pio/Meshtastic-esp32/src/main.cpp:319
0x400e799d: loopTask(void*) at /home/scott/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
0x400913c9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
scott@n550jk:~/pio/Meshtastic-esp32$ bin/exception_decoder.py -p ESP32 -e .pio/build/tbeam/firmware.elf /tmp/crash2
stack:
0x400d39d5: MeshService::loop() at /home/scott/pio/Meshtastic-esp32/src/MeshService.cpp:204
0x400d5ff3: loop() at /home/scott/pio/Meshtastic-esp32/src/main.cpp:319
0x400e799d: loopTask(void*) at /home/scott/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
0x400913c9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
scott@n550jk:~/pio/Meshtastic-esp32$ cat /tmp/crash1 /tmp/crash2
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x3ffe9300  PS      : 0x00060330  A0      : 0x800d39d8  A1      : 0x3ffd8ce0  
A2      : 0x3ffc54c0  A3      : 0x3ffe7d08  A4      : 0x00000005  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffd6e00  A8      : 0x800d39c4  A9      : 0x3ffd8cb0  
A10     : 0x3ffe91d8  A11     : 0x00000005  A12     : 0x00000000  A13     : 0x00000020  
A14     : 0x00060920  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x3ffe9300  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x3ffe9300:0x3ffd8ce0 0x400d39d5:0x3ffd8d00 0x400d5ff3:0x3ffd8d20 0x400e799d:0x3ffd8d50 0x400913c9:0x3ffd8d70
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x3ffe9238  PS      : 0x00060330  A0      : 0x800d39d8  A1      : 0x3ffd8ce0  
A2      : 0x3ffc54c0  A3      : 0x3ffe8bf4  A4      : 0x00000005  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffd6e00  A8      : 0x800d39c4  A9      : 0x3ffd8cb0  
A10     : 0x3ffe9238  A11     : 0x00000005  A12     : 0x00000001  A13     : 0x0000001c  
A14     : 0x3ffc52c8  A15     : 0x3ffd8c7c  SAR     : 0x0000000e  EXCCAUSE: 0x00000014  
EXCVADDR: 0x3ffe9238  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x3ffe9238:0x3ffd8ce0 0x400d39d5:0x3ffd8d00 0x400d5ff3:0x3ffd8d20 0x400e799d:0x3ffd8d50 0x400913c9:0x3ffd8d70
scott@n550jk:~/pio/Meshtastic-esp32$ 
```

I triggered it again a few times both nodes had  ```canSleep, mode=4, isRx=0, txEmpty=1, txGood=3``` repeating on the debug output and I woke them both up with button press, and one node had similar crash addresses to the above when sending packet to the node - maybe related to new receive collision avoidance? Apologies if this is expected, or due to my environment, I know you are doing a lot of great work and the normal release mechanism is via the zipfile. I thought I would add this in case you hadn't seen it.

geeksville, 2020-04-18:
Thank you for the complete and throughtful note.  Alas, my current code has diverged enough that I think line 204 in MeshService doesn't tell me enough.  

Btw - I did just check in a change to not print the "canSleep" message in the common case of "sleep is actually okay".  Instead I now only print if the radio is saying it needs to wait to finish some stuff.  (Related to bug #85)

closed 2020-04-18 by geeksville

spattinson, 2020-04-18:
I thought that may be the case, thanks