meshtastic/firmware #102: LG v20  unable to connect to radio

b3hr3ns, 2020-04-20:
I may have hit the report bug button a couple of times with this one as I wanted to make sure debugging info was there.  Was attempting to connect to heltec *6e:84 from my LG v20, which also has a crash or two (hope that helps in the ADB logs that are attached). [log2.txt](https://github.com/meshtastic/Meshtastic-esp32/files/4502483/log2.txt)

Also, with 0.4.3, I noticed my radios reset fairly often (white light goes solid) and then starts pinging like normal.

geeksville, 2020-04-21:
oh yes - thank you so much.  I made a fix for the problem you reported.
https://meshtastic.discourse.group/t/android-app-new-releases/78/19?u=geeksville

geeksville, 2020-04-21:
would you mind trying this new release and let me know if it works for you?  I'm happy to debug more if not (just click the report bug button again).

b3hr3ns, 2020-04-21:
I've now updated and the phone appears to connect, but sits at "looking for meshtastic devices" while the radio button for the connected device is selected.  When the app is in the device menu, and I switch to the message window, the app crashes.  The most recent report from the v20 should hopefully have this debug data.  

App ver: 0.4.8
heltec FW ver: 0.4.3 alpha

geeksville, 2020-04-21:
thanks.  hmm.  I'll look into the log tonight.  pasting here for now:

```
# Crashlytics - Custom logs
# Application: com.geeksville.mesh
# Platform: android
# Version: 0.4.8 (148)
# Issue: 4146b17b5dd15fecd176f17bf9dddfeb
# Session: 5E9E6E41009F000153865D28A95E9CE5_DNE_0_v2
# Date: Mon Apr 20 2020 22:42:00 GMT-0700 (Pacific Daylight Time)

  0 | Mon Apr 20 2020 20:53:37 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.MeshService: Creating mesh service
  1 | Mon Apr 20 2020 20:53:37 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.RadioInterfaceService: Creating radio interface service.  device=24:6F:28:AF:70:4E
  2 | Mon Apr 20 2020 20:53:37 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: Enqueuing work: connect
  3 | Mon Apr 20 2020 20:53:37 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth$BluetoothContinuation: Starting work: connect
  4 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.model.UIViewModel: ViewModel created
  5 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.MainActivity: Checking permissions
  6 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.MeshUtilApplication: Testlab is unset
  7 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.MainActivity: Binding to mesh service!
  8 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | screen_view
  9 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.analytics.GoogleAnalytics: Analytics: start screen Messages
 10 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.MeshService: in connectionState=DISCONNECTED
 11 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.MeshService: in getOnline, count=0
 12 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.MainActivity: Ignoring failure to get myId, service is probably just uninited... null
 13 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.MainActivity: connchange DISCONNECTED
 14 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.MainActivity$mesh$1: connected to mesh service, isConnected=DISCONNECTED
 15 | Mon Apr 20 2020 20:53:38 GMT-0700 (Pacific Daylight Time) | screen_view
 16 | Mon Apr 20 2020 20:53:41 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: new bluetooth connection state 2, status 0
 17 | Mon Apr 20 2020 20:53:41 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: work connect is completed, resuming status=0, res=kotlin.Unit
 18 | Mon Apr 20 2020 20:53:41 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.RadioInterfaceService: Connected to radio!
 19 | Mon Apr 20 2020 20:53:41 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: Enqueuing work: discover
 20 | Mon Apr 20 2020 20:53:41 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth$BluetoothContinuation: Starting work: discover
 21 | Mon Apr 20 2020 20:53:43 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: work discover is completed, resuming status=0, res=kotlin.Unit
 22 | Mon Apr 20 2020 20:53:43 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.RadioInterfaceService: Discovered services! Service size=7
 23 | Mon Apr 20 2020 20:53:43 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: Enqueuing work: reqMtu
 24 | Mon Apr 20 2020 20:53:43 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth$BluetoothContinuation: Starting work: reqMtu
 25 | Mon Apr 20 2020 20:53:43 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: work reqMtu is completed, resuming status=0, res=kotlin.Unit
 26 | Mon Apr 20 2020 20:53:43 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.RadioInterfaceService: requested MTU result=Success(kotlin.Unit)
 27 | Mon Apr 20 2020 20:53:44 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: starting setNotify(ed9da18c-a800-4f66-a670-aa7547e34453, true)
 28 | Mon Apr 20 2020 20:53:44 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: Enqueuing work: writeD
 29 | Mon Apr 20 2020 20:53:44 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth$BluetoothContinuation: Starting work: writeD
 30 | Mon Apr 20 2020 20:53:44 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.RadioInterfaceService: Broadcasting connection=true
 31 | Mon Apr 20 2020 20:53:44 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: Enqueuing work: readC 8ba2bcc2-ee02-4a55-a531-c525c5e454d5
 32 | Mon Apr 20 2020 20:53:44 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.MeshService: Received broadcast com.geeksville.mesh.CONNECT_CHANGED
 33 | Mon Apr 20 2020 20:53:44 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.MeshService: onConnectionChanged=CONNECTED
 34 | Mon Apr 20 2020 20:53:44 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: Enqueuing work: readC ea9f3f82-8dc4-4733-9452-1f6da28892a2
[... cut ...]
198 | Mon Apr 20 2020 22:41:49 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: starting scan
199 | Mon Apr 20 2020 22:41:49 GMT-0700 (Pacific Daylight Time) | com.geeksville.analytics.GoogleAnalytics: Analytics: start screen Settings
200 | Mon Apr 20 2020 22:41:49 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: BTScan component active
201 | Mon Apr 20 2020 22:41:49 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: scan already running
202 | Mon Apr 20 2020 22:41:49 GMT-0700 (Pacific Daylight Time) | screen_view
```

geeksville, 2020-04-21:
hmm - this look looks basically okay to me.  Though I did notice a mesage I expected to see missing from the log, which led to this discovery: https://github.com/meshtastic/geeksville-androidlib/commit/a449f224c3a7c76dc20accfa850a1e3555211b66

It crashed when you went to the messages screen?

geeksville, 2020-04-22:
Ok - I think this is fixed now.  Thanks again for the report!  Can you reopen if it still seems bad for you?

closed 2020-04-22 by geeksville