meshtastic/firmware #106: Unable to pair bluetooth using Android phones

android606, 2020-04-24:
Using both a (reasonably new) Motorola Moto G7 with Android 9, and a (super old) HTC One M7 with Android 4.4.

I tap "pair" on the phone, then sometimes get no response at all, other times get a prompt on the phone for a passkey, but get no key displayed on the device or serial port.

Also noticed the "signal xx seconds ago" display often resets to 0 at the instant you tap "pair".

geeksville, 2020-04-24:
hmm.  Would you be willing to attempt this again and then after it fails, go to the "report bug" button on the settings screen and press that?  Then comment here with the phone model?  That will allow me to get a full set of logs which will help track this down. 

Thanks for the report though.

android606, 2020-04-24:
Of course, no problem. I just did that, the device is "Android606's Moto G7 power".

If there are any logs you want me to pull from the phone, just let me know.