meshtastic/firmware #107: PeriodicScheduler should use a heap

geeksville, 2020-04-25:
fixme comment from PerodicTask.h:

class PeriodicScheduler
{
    friend class PeriodicTask;

    /**
     * This really should be some form of heap, and when the period gets changed on a task it should get
     * rescheduled in that heap.  Currently it is just a dumb array and everytime we run loop() we check
     * _every_ tasks.  If it was a heap we'd only have to check the first task.
     */
    std::vector<PeriodicTask *> tasks;
}

geeksville, 2020-04-25:
possibly an ordered stl::set

sachaw, 2022-05-12:
@mc-hamster Fairly certain this is no longer applicable right?

caveman99, 2022-05-16:
PeriodicTask is not in the firmware any more, i think (without looking) this was replaced by OSThreads...

closed 2022-05-17 by sachaw