meshtastic/firmware #116: device code fails to pair with a moto G6 or HTC One M7

android606, 2020-04-29:
Using both a (reasonably new) Motorola Moto G7 with Android 9, and a (super old) HTC One M7 with Android 4.4.

I tap "pair" on the phone, then sometimes get no response at all, other times get a prompt on the phone for a passkey, but get no key displayed on the device or serial port.

Also noticed the "signal xx seconds ago" display often resets to 0 at the instant you tap "pair".

android606, 2020-05-01:
I just downloaded / compiled / flashed the latest code from master a few minutes ago, and I'm now unable to reproduce this. The PIN code is displaying, Moto G7 is pairing, stuff is working!

I have no idea what changed, but this issue seems to be fixed.

geeksville, 2020-05-03:
Interesting!  Good news I guess.  I have a theory though and I'd like to ask you: When you flashed it originally did you flash it using my prebuilt bins and the esptool or did you flash it using the platform io tool?  The ESP32 BT stack keeps some state in the flash filesystem and I'm beginning to suspect that esptool is missing some bit of init that the platformio tool does...

android606, 2020-05-04:
Hmm... Good theory! The problematic FW was prebuilt / esptool, and the one that's working now was built and flashed using VS Code / platformio.

spattinson, 2020-05-23:
I have found that to resolve pairing issues you have to force close meshastic from android settings -> apps menu - I can quickly get to that by long pressing the icon in my launcher (Nova) and selecting app info. Closing it from your launcher task list does not actually close it, its in exactly the same state when you launch it again - at least on my phone. After you have FC'd it you can get it to attempt pairing again instead of being stuck. You do also need to delete the pairing in BT settings too. As far as the pairings on the esp32 go, sometimes they can be preserved between flashing if you dont do a erase flash, other times not, and other times you need to erase, or at least in previous versions you did, pairing seems to be more reliable than earlier versions of meshtastic.

geeksville, 2020-06-17:
Sounds like our custom installer script fixed this one.  Closing until there are new reports.

closed 2020-06-17 by geeksville