meshtastic/firmware #130: FW 0.6.3 oled is black on T-Beam v1.0

claesg, 2020-05-08:
Downloaded and flashed FW 0.6.3 on two T-Beam V1.0 EU version. 

./device-install.sh firmware-ttgo-lora32-v1-EU865-0.6.3.bin

After the reboot the oled screen is black. Same on both devices. Had to look in the serial monitor to get the BT code to be able to pair. I had some problem to get them paired but eventually I managed. Sending msgs between phones works ok.

log from device:
--
booted, wake cause 0 (boot count 1), reset_reason=reset
No I2C devices found
Meshtastic swver=0.6.3, hwver=1.0-EU865
Read RTC time as 0 (cur millis 54) valid=0
ERROR: No UBLOX GPS found, hoping that NEMA might work
RadioConfig reset!
Loading saved preferences
Warn: devicestate is old, discarding
NODENUM=0xa0, dbsize=1
Starting meshradio init...
Set radio: name=Default, config=3, ch=0, txpower=23
LORA init result 0
sending owner !246f289794a0/Unknown 94a0/?A0
Update DB node 0xa0, rx_time=0
old user !246f289794a0/Unknown 94a0/?A0
updating changed=0 user !246f289794a0/Unknown 94a0/?A0
Adding broadcast record for fr=0xa0,to=0xff,id=232
enqueuing for send on mesh fr=0xa0,to=0xff,id=232 (txGood=0,rxGood=0,rxBad=0)
pressing
Trigger powerFSM 1
Starting low level send from=0xa0, id=232!
Transition powerFSM transition=boot timeout, from=BOOT to=ON
Setting bluetooth enable=1
Pre BT: 182332 heap size
Starting bluetooth
Completed sending to=0xff, id=232
*** Mesh service:
Lora RX interrupt from=0xa0, id=232
Found existing broadcast record for fr=0xa0,to=0xff,id=232
Ignoring incoming floodmsg, because we've already seen it
Lora RX interrupt from=0x14, id=199
Notifying observers of received packet fr=0x14,to=0xa0,id=199
Trigger powerFSM 3
Ignoring incoming packet - not a position
Forwarding to phone, from=0x14, rx_time=0
Update DB node 0x14, rx_time=0
old user //
updating changed=1 user !246f28979a14/Nexus6/N
Trigger powerFSM 8
Transition powerFSM transition=NodeDB update, from=ON to=ON
Telling client we have new packets 1
Telling client we have new packets 1
onPassKeyNotify 767902
Trigger powerFSM 7
Transition powerFSM transition=Bluetooth pairing, from=ON to=ON
 authenticated and connected to phone
Lora RX interrupt from=0x14, id=201
Adding broadcast record for fr=0x14,to=0xff,id=201
Rebroadcasting received floodmsg to neighbors, fr=0x14,to=0xff,id=201
enqueuing for send on mesh fr=0x14,to=0xff,id=201 (txGood=1,rxGood=3,rxBad=0)
Notifying observers of received packet fr=0x14,to=0xff,id=201
Trigger powerFSM 3
handled incoming position time=0
Forwarding to phone, from=0x14, rx_time=0
Update DB node 0x14, rx_time=0
Sending network ping to 0x14, with position=0, wantReplies=0
sending owner !246f289794a0/Unknown 94a0/?A0
Update DB node 0xa0, rx_time=0
old user !246f289794a0/Unknown 94a0/?A0
updating changed=0 user !246f289794a0/Unknown 94a0/?A0
enqueuing for send on mesh fr=0xa0,to=0x14,id=233 (txGood=1,rxGood=3,rxBad=0)
Telling client we have new packets 2
Telling client we have new packets 2
Can not send yet, busyRx
Lora RX interrupt from=0x14, id=200
Adding broadcast record for fr=0x14,to=0xff,id=200
Rebroadcasting received floodmsg to neighbors, fr=0x14,to=0xff,id=200
enqueuing for send on mesh fr=0x14,to=0xff,id=200 (txGood=1,rxGood=4,rxBad=0)
Notifying observers of received packet fr=0x14,to=0xff,id=200
Trigger powerFSM 3
Ignoring incoming packet - not a position
Forwarding to phone, from=0x14, rx_time=0
Update DB node 0x14, rx_time=0
Received text msg from=0x14, id=200, msg=zzz
Trigger powerFSM 5
Transition powerFSM transition=Received text, from=ON to=ON
Sending network ping to 0xff, with position=0, wantReplies=1
sending owner !246f289794a0/UnknStarting low level send from=0x14, id=201!
own 94a0/?A0
Update DB node 0xa0, rx_time=0
old user !246f289794a0/Unknown 94a0/?A0
updating changed=0 user !246f289794a0/Unknown 94a0/?A0
Adding broadcast record for fr=0xa0,to=0xff,id=234
enqueuing for send on mesh fr=0xa0,to=0xff,id=234 (txGood=1,rxGood=4,rxBad=0)
Telling client we have new packets 3
Telling client we have new packets 3
Can not send yet, busyTx
Can not send yet, busyTx
Completed sending to=0xff, id=201
Starting low level send from=0xa0, id=233!
Completed sending to=0x14, id=233
Starting low level send from=0x14, id=200!
Completed sending to=0xff, id=200
Starting low level send from=0xa0, id=234!
Completed sending to=0xff, id=234
Lora RX interrupt from=0xa0, id=234
Found existing broadcast record for fr=0xa0,to=0xff,id=234
Ignoring incoming floodmsg, because we've already seen it
Lora RX interrupt from=0x14, id=202
Notifying observers of received packet fr=0x14,to=0xa0,id=202
Trigger powerFSM 3
Ignoring incoming packet - not a position
Forwarding to phone, from=0x14, rx_time=0
Update DB node 0x14, rx_time=0
old user !246f28979a14/Nexus6/N
updating changed=0 user !246f28979a14/Nexus6/N
Telling client we have new packets 4
Telling client we have new packets 4
Lora RX interrupt from=0x14, id=203
Notifying observers of received packet fr=0x14,to=0xa0,id=203
Trigger powerFSM 3
handled incoming position time=0
Forwarding to phone, from=0x14, rx_time=0
Update DB node 0x14, rx_time=0
Telling client we have new packets 5
Telling client we have new packets 5
 authenticated and connected to phone
Trigger powerFSM 9
encoding toPhone packet to phone variant=2, 53 bytes
Trigger powerFSM 9

--

claesg, 2020-05-08:
Tried to go back to a homebuilt 0.6.2 and the screen is still black which worked fine before I flashed 0.6.3.

geeksville, 2020-05-08:
Hi @claesg,

It looks like you are using the wrong board ID:

You have "./device-install.sh firmware-ttgo-lora32-v1-EU865-0.6.3.bin"

If you change it to "./device-install.sh firmware-tbeam-EU865-0.6.3.bin" I think you'll probably be fine.

claesg, 2020-05-08:
Yes. You are right. I am sorry for my stupid mistake. Using the correct FW is much better :-) Now everything works.

closed 2020-05-08 by claesg