meshtastic/firmware #133: force deep-sleep state if battery falls below 15% full

geeksville, 2020-05-11:
To prevent battery damage, if we ever see battery voltage getting below 3.7Vish (see https://blog.ampow.com/lipo-voltage-chart/) we should have PowerFSM.cpp transition the device to deep-sleep state.

spattinson, 2020-05-13:
axp192 already has this built in, rather than deep sleep it cuts off power to esp32, lora, gps, and probably display? There is a configurable cut off voltage. I think I mentioned before that it is somewhat on the low side for protecting the battery. I'll have a look into it, It'll be a matter of setting it at initialisation then it should take care of it automatically and turn back on when you charge.

geeksville, 2020-05-13:
ooh that sounds even better - I had forgotten that.  If you could do that then at would be great.  yes - to do this we would just need to trigger a (new to define (LOWBATTERY in the list of triggers and then have that irq handler send that trigger event.  Then PowerFSM.cpp could have a new transition added that goes into deep sleep any time that event occurs.

commit efc23953 (2020-05-19): Fix #133 - force deep sleep if battery reaches 10%
closed 2020-05-19 by geeksville