meshtastic/firmware #134: Updates to the display/add icons

mehuman, 2020-05-12:
Update the usability and information on the default display. This may have initially been meant to be a debug screen but I find the information useful and think it should be promoted to a first class screen.

Following the suggestion of the existing [TODO](https://github.com/meshtastic/Meshtastic-device/blob/master/src/screen.cpp#L646) add icons to the display.

Icons added/to be added.
Power
  - Battery
  - USB
  - Calculate battery percentage?
GPS
  - Check mark for good
  - X for bad

Other informative lines to add:
  - Time?
  - Elevation?

I've started work in a forked branch. Currently there are stubbed out icons for USB and Battery. I've removed the `drawRows` function to allow proper formatting with drawing the icons. The idea is now to draw the left side of the screen and then switch the text justification to the right and draw that side of the screen.

[WIP Branch](https://github.com/mehuman/Meshtastic-device/tree/icon-display)

Any and all help, suggestions, comments welcome

geeksville, 2020-05-12:
wow - sounds great.  yes, the main reason I called it a "debug" screen was that the current version is so cheezy ;-).  If you want to make it actually quality then I think that's a great idea.

I haven't looked at your branch yet but if it helps there are some scripts in /bin that I used to generate the existing xbm .h files.  Also, the material.io icons (and the various forks of that icon set) was what I have been using thus far (but it sounds like you have better ideas in that area)

mehuman, 2020-05-12:
I use material.io icons already with Homeassistant and was exactly the thought. Didn't see the scripts, I'll check them out.

geeksville, 2020-05-13:
oops my memory was wrong.  the script is in /images/genfile.sh.  If you add new svgs feel free to update that script (or not)

geeksville, 2020-06-12:
@mehuman Just checking in - any progress on this?  (no pressure)

Professr, 2020-06-20:
I started playing with some icons before I checked the issues, but if you're interested in using them...
`static const uint8_t imgUSB[16] =           { 0x60, 0x60, 0x30, 0x18, 0x18, 0x18, 0x24, 0x42, 0x42, 0x42, 0x42, 0x7E, 0x24, 0x24, 0x24, 0x3C };
static const uint8_t imgPower[16] =         { 0x40, 0x40, 0x40, 0x58, 0x48, 0x08, 0x08, 0x08, 0x1C, 0x22, 0x22, 0x41, 0x7F, 0x22, 0x22, 0x22 };
uint8_t imgBattery[16] =                    { 0xFF, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0xE7, 0x3C };
static const uint8_t imgUser[8] =           { 0x3C, 0x42, 0x99, 0xA5, 0xA5, 0x99, 0x42, 0x3C };
static const uint8_t imgPositionEmpty[6] =  { 0x20, 0x30, 0x28, 0x24, 0x42, 0xFF };
static const uint8_t imgPositionSolid[6] =  { 0x20, 0x30, 0x38, 0x3C, 0x7E, 0xFF };
`
[image]
[image]

geeksville, 2020-06-21:
ooh that's awesome.  It might be that @mehuman never got back to this.  @mehuman are you still working on this?  If not, it sounds like @Professr might be willing.

geeksville, 2020-06-21:
@Professr alas, it seems like mehuman might have moved on to other interests (temporarily)?  If you'd like to send in a PR with your slick new GUI that would be awesome.

geeksville, 2020-06-21:
or you could pull from @mehuman's branch above as a starting point?

Professr, 2020-06-21:
I'll take a look :)

Professr, 2020-06-22:
The branch is at the point where there are XBM icons replacing some of the symbols.  I'd be interested in seeing if those images are preferable to my hand-drawn ones, but I'm using drawFastImage instead of drawXbm so they're not directly compatible.  I could always convert them if so.
I have the code tested and ready, but it relies on the branch that adds battery percentage (which I already sent in a PR for).  Uhh, how should I handle that?

Professr, 2020-06-23:
Merged in, feel free to add / suggest any changes.  If there aren't any, can close the issue if/when you want.

geeksville, 2020-06-23:
Thanks @mehuman for the initial idea and thanks @Professr for doing all this!

closed 2020-06-23 by geeksville

mehuman, 2020-10-02:
Totally missed seeing this during a move. Yes, thank you for all the work!