meshtastic/firmware #137: How to get the BT pin?

vaderj, 2020-05-16:
OS: Win 10 1909
Using esphome-flash-1.2.0-windows-x64.exe, after completing the writing of : 

firmware-ttgo-lora32-v2-US-0.6.3.bin 

to a TTGO T-Beam, I receive the following output (after clicking "View Logs")

[image]

Am I looking in the wrong spot? I looked through the forum and I didn't see anything that looked related

closed 2020-05-16 by vaderj

vaderj, 2020-05-16:
I realized that I had not written the correct firmware, but even after writing the tbeam firmware, the BT Pin is NOT printed to the diag output that is visible from the ESPHome util.

I tried looking at the port via Arduino Serial Monitor but none of the speeds resulted in readable text.

I was just trying to verify the hardware booted before I soldered stuff up, and just applying manual pressure with the screen allowed me to see the Bluetooth Pin and verify the devices, at least BT and firmware, worked.

Thats for such an awesome project! I hope to be able to contribute in some form soon!

geeksville, 2020-05-16:
Hmm - based on your original screencap, it looks like you might be talking
to the device at 115200 baud.  Try 921600 - that's the baud rate our
appload uses - I bet you'll see a lot of messages.

vaderj, 2020-05-17:
I don't see a way to change the write speed in the ESPHome-Flasher in Windows,