meshtastic/firmware #140: Export GPS History

paulcart1, 2020-05-19:
As the project stabilizes it would be great to add the following feature:

Add the ability to export the GPS history of a node (GPX or similar format) so as to be able to review routes at a later date. 

- [ ] Ability to extract historic GPS data from the app

- [ ] Units not connected to the app should be able to send regular location updates to be captured by a node in the network that is connected to the app or node with SD card. While some update may be missed hopefully enough will be captured to recreate a high percentage of the route taken.

maufl, 2020-06-02:
Hey, I'm interested in using a meshtastic device as a GPS logger. This issue sounds like the GPS coordinates would be sent to the mobile phone which would store it. Would it also be possible to store the waypoints on the device and only periodically sync them?

geeksville, 2020-06-02:
@maufl yes, that would be a fairly painless feature to add.  I'm happy to give tips/advice to anyone who wants to add it, but I personally won't have time to work on it for about a month.  I'm working on getting the current release to '1.0' quality with the existing features and the (few) remaining TODOs.

slavino, 2020-06-30:
The TTGO LoRa32 v2.1 (and few others) come with SD card adapter.

So after boot if SD card is present and writeable, meshtastic could wait for valid GPS data and log it based on "some" interval
the file format for rolling file could be "YYYY_mm_dd.gpx" in GPX format (https://www.topografix.com/gpx.asp) which is one of the standards supported across many vendors.

The boards not equipped with SD card slot may benefit from OpenLog project (https://github.com/sparkfun/OpenLog) in case of  interest from user's side.

slavino, 2020-07-15:
@maufl please take a step back and agree at least with @geeksville , @Professr and maybe few others on the design of the storage (most probably SD card) implementation.
There do exist options:
- on board SD card holder accessible via SPI
- additional SPI SD card board (similar approach used to add GPS capability to TTGO LoRa32 boards)
- additional module like OpenLog that only requires Software Serial
- etc.

*So some level abstraction might make sense as well here otherwise there will be lots of refactoring needed just as it is in case of battery and it's Power.cpp/Power.h currently on TTGO T-Beam v0.7 versus v1.0/v1.1 where AXP192 chip is missing.*

Professr, 2020-07-15:
Yes, we will need to be able to detect the presence of a generic file storage device (possibly multiple storage devices) and have abstract methods for reading / writing just as we will for user input devices :\  I can think of several other intended projects that will need storage handling, so this will be a very cool feature to have

aHVzY2g, 2020-07-17:
don't forget the SPIFF, depending on the update rate it's doesn't affect the lifetime of it.

mc-hamster, 2021-03-12:
Closing this ticket. The function described here can be done in the range test plug-in.

closed 2021-03-12 by mc-hamster