meshtastic/firmware #143: Support PineTab?

poVoq, 2020-05-22:
The 100$ Linux tablet PineTab will come with an optional LoRA adapter:
https://www.pine64.org/2020/05/15/may-update-pinetab-pre-orders-pinephone-qi-charging-more/

Might be a nice all in one device, although porting the Android app to qt Linux might not be so strait forward.

android606, 2020-05-22:
Maybe a cross-platform Qt Linux and iOS app would kill two birds with 1-1/2 stones?

poVoq, 2020-05-22:
Or maybe a Flutter port of the Android app would be possible? That way you could even have a webversion.

geeksville, 2020-05-22:
oh this piece of kit sounds awesome.  And yes, I've also been recently pondering  cross platform iOS/Linux frameworks (btw @jeksys how's iOS going with your app?)

jeksys, 2020-05-22:
@geeksville Slower than I want it. But still working on it from time to time. I was actually considering to use Flutter at some point. But I want finish native first. 
Actually Flutter has BLE plugin, interesting.

geeksville, 2020-05-22:
okay @jeksys no worries - just checking in.  If I can help in any way please let me know.

@poVoq thanks for posting this, I had never heard of this cool project. FWIW, I just sent them an email and asked about getting a development build of their lora board once they have it.

geeksville, 2020-06-17:
Btw - I just received an email from the Pinetab folks, they are sending me an early hardware build and I'll try to add some meshtastic support to it.

happytm, 2020-06-17:
@poVoq  nice find. Thanks.

geeksville, 2020-06-18:
thanks for mentioning it!