meshtastic/firmware #154: utf8 strings look bad on the OLED screen

geeksville, 2020-06-04:
We should make a fixupUTF8() function and use it in screen.cpp when we print strings.  Substituting ? for utf8 characters.

Professr, 2020-06-22:
It looks like the OLED library does support utf8 that maps to extended ASCII, just not all non-latin characters.  The default conversion function just returns a NUL if a char isn't convertable, so I made one that returns a '?' instead.  I'm not really set up to test non-EASCII utf8 though, so I can't be sure it does what we want :\

geeksville, 2020-06-22:
that sounds great, I'd recommend testing by copying and pasting some japanese or chinese into the android app and seeing it it properly gets ? motized.  The for the extended uth8 encodings is just to look at the first byte and based on the upper few bits you know how many bytes you need to skip.  Table here: https://en.wikipedia.org/wiki/UTF-8#Description

Professr, 2020-06-22:
I don't actually have an android device yet; I've only sent a few test messages via the python API (which would probably work for testing this too I suppose).  If anyone else feels like giving a whirl, go for it, otherwise I'll test it at some point since this isn't critical yet

aHVzY2g, 2020-06-23:
I did some testing, for me it works. I would recommend we use a ¿ instead of ?, it would make it easier to find your answer as a user if you search "meshtatic ¿" . Also, it may be better to reduce the output to a single "¿". For example "的" returns "???"

Professr, 2020-06-24:
Well that was interesting; the font map we're using doesn't seem to support the euro symbol, and since it doesn't follow the standard EASCII I'm used to from ASCII tables it took a little while to find the right character code for ¿ (the UTF-8 conversion function goes byte by byte, and extended characters are two bytes long - so I couldn't just return '¿', I had to find the code it mapped to).
tl;dr: I was able to make the changes you suggested

aHVzY2g, 2020-06-24:
just tested again, works great 👍

commit caafddfd (2020-06-24): Merge pull request #219 from Professr/issue#154

Fixed #154 - Cleaned up UTF-8 conversion, tweaked screen.cpp to replace a hardcoded constant
closed 2020-06-24 by geeksville