meshtastic/firmware #162: Use spare TBEAM gpios to provide alternate "power" and "next" buttons

geeksville, 2020-06-07:
per @sexycyborg:
> is there any way two of the spare pins could be used for the momentary power switch? It would make enclosure designs a lot easier if we didn’t have to work with those tiny SMD buttons and could wire up proper waterproof ones.

geeksville, 2020-06-12:
Hi @sexycyborg,

I'm going through my work queue now. You requested alternate GPIOs for the TBEAM buttons. I think doing this for the middle button will be easy. Also the reset button is also easy (already there). But the power button would be hard (not controllable by sw - goes straight to the PMU).

I assume you really only wanted the middle button right? because the main power button (as long as deep sleep works - which seems solid) should only need to be pressed once.

geeksville, 2020-06-17:
sounds not urgent now, so I'm moving to the post 1.0 bucket for now.  If useful soon, just ping me and it is easy.

Professr, 2020-06-22:
Looks like the plan was to use OneButton for debounce, so I migrated the button press code and added a BUTTON_PIN_ALT set to pin 13.  It seems to run at least as smoothly as it did before, and shorting 13 to GND now triggers the same action as the middle button.  Thoughts on using that particular GPIO?

geeksville, 2020-06-23:
fixed by @professr!

closed 2020-06-23 by geeksville