meshtastic/firmware #167: wait to sleep bug

geeksville, 2020-06-09:
```
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
NOTE! Recording critical error 2, address=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
```

geeksville, 2020-06-10:
Oh!  Root cause is that the wait to sleep loop is consuming the main thread - and therefore starving the handler that would have sent the message (see the last "delaying " line in this smoking gun trace):

```
Telling client we have new packets 3
Telling client we have new packets 3
Starting low level send from=0xabdddf38, to=0xffffffff, id=250959435, want_ack=0
Completed sending to=0xffffffff, id=250959435
tx complete - starting timer
processUBX: counter hit MAX_PAYLOAD_SIZE
new gps pos lat=0.000000, lon=0.000000, alt=0
processUBX: counter hit MAX_PAYLOAD_SIZE
Ignoring invalid GPS time
processUBX: counter hit MAX_PAYLOAD_SIZE
Ignoring invalid GPS time
processUBX: counter hit MAX_PAYLOAD_SIZE
new gps pos lat=0.000000, lon=0.000000, alt=0
Ignoring invalid GPS time
processUBX: counter hit MAX_PAYLOAD_SIZE
Transition powerFSM transition=Screen-on timeout, from=ON to=DARK
Turning off screen
processUBX: counter hit MAX_PAYLOAD_SIZE
Ignoring invalid GPS time
processUBX: counter hit MAX_PAYLOAD_SIZE
new gps pos lat=0.000000, lon=0.000000, alt=0
Ignoring invalid GPS time
Transition powerFSM transition=Bluetooth timeout, from=DARK to=LS
lsEnter begin, ls_secs=3600
lsEnter end
lsIdle begin ls_secs=3600
Setting bluetooth enable=0
Shutdown BT: 174816 heap size
reached ls_secs, servicing loop()
processUBX: counter hit MAX_PAYLOAD_SIZE
sending owner !2462abf84098/Unknown 4098/?98
Update DB node 0xabf84098, rx_time=0
old user !2462abf84098/Unknown 4098/?98
updating changed=0 user !2462abf84098/Unknown 4098/?98
Adding packet record for fr=0xabf84098,to=0xffffffff,id=293394629
enqueuing for send on mesh fr=0xabf84098,to=0xffffffff,id=293394629 (txGood=5,rxGood=3,rxBad=0)
delaying 1354
lsIdle begin ls_secs=3600
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
radio wait to sleep, txEmpty=0
```

slavino, 2020-06-10:
https://www.thethingsnetwork.org/forum/t/ttgo-t-beam-v1-0-oled-deep-sleep-needs-proper-power-design/30800/11

todd-herbert, 2024-03-27:
Still seeing this four years later, on Heltec Wireless Paper