meshtastic/firmware #169: Home Assistant Integration

rradar, 2020-06-11:
It would be a great communication channel. If integrated in [Home Assistant](https://www.home-assistant.io/) it could be used with `notify:` and take immediately advantage of all powers home assistant offers.

For example I could text `weather` and home assistant can be configured (conversation/intents) to actually reply with the actual weather forecast information I set. :partly_sunny:

If I'm right the linux application is written in python already (just like home assistant) so the work is maybe already half done for this :grin:

mehuman, 2020-06-12:
Hi fellow HA user,
  I've brought this up on the forums and have, at least I think, an interesting use case in smarthomeing a small RV and using meshtastic as a source of communication when in the backcountry.

MQTT support is planned [link](https://meshtastic.discourse.group/t/next-work-items-in-my-queue/153) which should get Meshtastic and HA talking. Something beyond that might be out of scope for a project this new, unless someone takes the initiative to code it. I was dreaming of an ESPHome/Meshtastic convergence so I can add modules to the device.

geeksville, 2020-06-12:
(Not exactly home assistant - but I'm going to repurpose this bug for tracking MQTT support)

rradar, 2020-06-12:
Despite MQTT (20 year old protocol by now) get's supported I would really like to see a **real/native support in home assistant**.

I finally turned off my mqtt broker last year and I could not think of a happier moment in my 'technical career'. It just sucks so much time to get server/clients speaking proper to each other and debugging mqtt in general is just a lot of a mess

geeksville, 2020-06-12:
I think native home assistant support would be great and would happily
accept PRs / help anyone who wants to do it.

slavino, 2020-06-12:
Hardwiring to any solution doesn't put this project any further. Providing a set of adapters is a much better approach as it literally opens gates for any integrator.

- Easy to use REST API can open gates even to cheap hosting of PHP based solutions.
- Simple MQTT integration can let it grow even further if there will be well maintained versioning of messages (because I believe it will pass JSON strings back and forth). And setting up MQTT broker on RaspberryPi is literally question of few minutes.

So rather than doing proprietary binding to HASS I do suggest to keep it simple and develop a DEMO for cheap and time-wise affordable solutions/integrations.  To setup HASS just to tryout Meshtastic capablities in terms of interaction with server backend seems a bit overkill to me...

geeksville, 2020-06-30:
possibly use protobufs as an alternative MQTT encoding, which would allow sending tiny ints rather than long topic strings.  Mapping from those ints to "standard" MQTT strings would happen automatically inside the node and the gateway.

Try to move as many of the existing services as possible (position and text messaging) to be built **on top of** this micro-MQTT implementation.

geeksville, 2020-07-13:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/rak815-can-connect-with-thingsboard-platform-using-lora-gateway/721/6

rradar, 2020-12-13:
I'm not a developer but from what I can tell there is a [python implementation for meshtastic](https://github.com/meshtastic/Meshtastic-python) (Python 97.5%) and there is the [Home Assistant Integrations](https://www.home-assistant.io/integrations) which are a [100% Python](https://github.com/home-assistant/core), maybe most of the work is actually already done? :thinking:

geeksville, 2021-01-28:
btw - design plan here: https://github.com/meshtastic/Meshtastic-device/blob/master/docs/software/mqtt.md

geeksville, 2021-05-28:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/mqtt-roadmap-and-release/3232/2

samuk, 2021-11-05:
Any updates on this one, would like to play with some sensors.

closed 2021-12-12 by sachaw

samuk, 2021-12-13:
Great to see this happening. Any idea when it will be in a release?

sachaw, 2021-12-13:
riot isn't implemented yet, still in 'feature request' stage, but mqtt can be used: https://meshtastic.org/docs/software/settings/mqtt