meshtastic/firmware #174: Serial monitor baud rate 115200 not possible

rradar, 2020-06-12:
I just tried again to confirm and cloned this repo freshly and edited the `platformio.ini` modifying to this values:

```yaml
monitor_speed = 115200
;monitor_speed = 921600
```
```yaml
upload_speed = 115200
;upload_speed = 921600
```

After compiling and uploading (target is heltec but the device is a esp32 lolin32 lite paired with a ra-02 module and a oled) I can't access serial in a proper manner:

> cu -l /dev/ttyUSB0 -s 115200
```
Connected.
�������������.Ɯ��;���ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
��
  ���Ɯ�Ϥ����F������)���d��!�^������ܔ��ԕ����Ŧ���X"�
                                                    ϼ���������(�Ƅ�|��`��������4���dΏ�d��aY��m��)�N��N�:��������������/�����
                                                                                                                                    ����� ��
```

Also using meshtastic python outputs just garbage:

> meshtastic

```
?????????^s?????)?liA?U??VK??
                              ???9?teA?V??*/??????	????:????:??????????????N?????????????nk??????TuA?E??nK??
                                                                                                                      ???-?huA?N??*7??????????:????:???????
```

rradar, 2020-06-12:
I tried exactly the same code on a Lolin32 (without lite) and ra-02 combination and serial monitor is still garbage:

> cu -l /dev/ttyUSB0 -s 115200
```
Connected.
������ǔ&|�Q�ñ��y)�ͽ�
```

but meshtastic python works here:

> meshtastic 
```
Trigger powerFSM 9
Trigger powerFSM 11
Trigger powerFSM 1
Connected to radio...
Transition powerFSM transition=boot timeout, from=BOOT to=ON
Pre BT: 189136 heap size
```

geeksville, 2020-06-12:
yes, our device software talks at 921600.  If you want to change this (for a particular device- if your serial part can't run that fast), change SERIAL_BAUD in configuration.h

Meshtastic python is working because it is talking to the devices at 921600.

geeksville, 2020-06-17:
Based on forum posts it sounds like @rradar is fine now.  Please reopen if this is still a problem.

closed 2020-06-17 by geeksville