meshtastic/firmware #177: Support Dragino Lora Gps Pi Hat

r0kk3rz, 2020-06-13:
I have a couple of these devices, and it would be nice to be able to use them with meshtastic, it might make a nice development platform

https://www.dragino.com/products/lora/item/106-lora-gps-hat.html

I'd be happy to take a look at this myself and any pointers on where to start would be great

geeksville, 2020-06-13:
Looks like a nice board.  I think mostly what you are describing is a port so this software can also run on a raspberry pi.  I'd start by looking around and seeing if anyone has already written an "arduino" clone API that sits on top of raspbian.   Because our code includes drivers for the SPI radio chip, that will be pretty different in linux land (i.e. ioctl calls etc instead of just touching registers).    So you'll want to find if someone has already done a library that does that (painful) work.

Porting from the land of RTOSes to running on top of linux as a userspace app will be biggish effort I expect.  But if you are hip to do it I think it is great.

you also might find this thread useful:

https://meshtastic.discourse.group/t/discussion-on-support-for-custom-build-targets-i-e-different-radio-chips-cpus-one-off-boards/104/38

r0kk3rz, 2020-06-13:
Yeah i figured it could be a tricky thing to achieve, and might be better off making a meshtastic compatible python implementation for this use instead

geeksville, 2020-06-13:
Another option you could consider: A $15-$30 (depending on options) board running meshtastic, plugged into the USB port on the Pi and use meshtastic-python.

geeksville, 2020-06-17:
yep - seems like a $15 meshtastic to USB board is a better way to go.  Closing for now

closed 2020-06-17 by geeksville