meshtastic/firmware #178: Add support for  SH1106 type OLED display

mrvdb, 2020-06-13:
I'm using an 1.3" oled display which has the SH1106 controller apparently.

I'm hit by this issue: https://luma-oled.readthedocs.io/en/latest/troubleshooting.html

Looking at the used library https://github.com/meshtastic/esp8266-oled-ssd1306.git  it looks like the SH1106 is supported.

mrvdb, 2020-06-13:
Turns out it's fairly trivial:
- use SH1106Wire instead of SSD1306Wire as `dispdev` object
- treat SH1106 the same as SSD1306 in submodule, meaning apply the change that allows -1 as value for _sda and _scl  and I have applied additions for `initI2cIfNeccesary`  

I'm not sure exactly how to proceed here. A change is involved to the submodule repository for the OLED display, which is already patched from the upstream library. For this repository, should the choice for the controller be defined in configuration.h? 

I have a working version with the above changes applied

geeksville, 2020-06-13:
Yes - I think adding something like the following to configuration.h would be great:

#define USE_SSD1306
#define USE_SH1106

and then put those #defines into the (nasty) series of board definitions.  Or if you want an easier merge how about:

initially define USD_SSD1306 in configuration.h and then in the #ifdef for your new board just undef that and instead definte USE_SS1106 (since all boards thus far have been SSD1306).

I'll be needing to do something similar when I add support for a new LCD based board and I'll try to make the three cases fairly clean.

Sound good?

mrvdb, 2020-06-13:
What do you mean by

> the (nasty) series of board definitions.

?

geeksville, 2020-06-13:
heh.  Mostly that I'm not proud of how the current configuration.h does the ifdef changes for the different possible board types.

This series  of if-then-else clauses:

https://github.com/meshtastic/Meshtastic-device/blob/master/src/configuration.h#L205

Although I just had a slightly different idea: If the SH1106 has a different I2C device address you could check for it dynamically, in the same way we currently probe for the SSD1306?

mrvdb, 2020-06-13:
As far as I know they have the same i2c address 0x3c (possibly 0x3d for 
dual  display use) so distinguishing the controller based on that isn't 
possible I think.

closed 2020-06-15 by geeksville