meshtastic/firmware #181: 0.7.5 beta

feh123, 2020-06-14:
Using a tbeam with EU865 0.7.5 I could not get the unit to pair with app 0.7.71, I tried going back to 0.7.4 but the app was still not detecting the devices. I reported a bug on the app.
Also 0.7.4 was not detecting my other tbeams running 0.6.8.
Do you need any other info from me? Can I downgrade the app to an earlier version to check if that fixes the problem? Thanks.

geeksville, 2020-06-14:
Can you try the app I released yesterday 0.7.72 - it fixes a big bluetooth
problem.  If that doesn't work, you can go to the "google play" app and
click to leave the beta test, then it will let you install the stable
version.  Thanks for the report!

geeksville, 2020-06-14:
btw - the reason I think 0.7.72 might help is: https://meshtastic.discourse.group/t/alpha-tester-thread-please-try-0-7-5-device-and-0-7-72-android/298/31?u=geeksville 

also, can you try rebooting your phone and tell me if the devices then show up?  (I forgot - what phone model do you have?)

feh123, 2020-06-14:
Thanks I will try 0.7.72 - this is a great project - I wish I could help more.

feh123, 2020-06-14:
Hi
It's a Sony Z1 with Android 5.1.1.
Will will try these checks tomorrow.

feh123, 2020-06-15:
Hi
Using 0.7.71 on the app one of my devices on 0.6.8 connected. I could write a message on the app and send it but it never arrived at the device. Also the meassage icon in the sent area said ?00 - the device is called 4ec0. I did send a report bug.

I will try to update to 0.7.72 now.
In fact updated to 0.7.73! This worked better and connected to the device on 0.7.4 but the three devices did not form a mesh. I reported this too.
Now my two 0.6.8 devices are meshed but the 0.7. is not part of it and the OLED says 1/1 users whereas the 0.6.8 devise have 1/2 users.
The number of devices seen on the app screen varies - they seem to drop off and then re-appear. I have tried several times but cannot text them.

I will update all my devices to 0.7.5

feh123, 2020-06-15:
Hi
I have updated all my tbeams to 0.7.5 beta. They all mesh now although all give the OLED screen message gps is bad. I have checked two of them by putty over serial and none have given a lat/lon position.

I am using 0.7.73 on the app - it has seen all the devices. It does not keep them all on screen however. They seem to appear and disappear randomly. I guess this is as they make contact with the app? None show a position in mapbox.
Just to say I have updated my devices via esptool.py - is this a good way?

geeksville, 2020-06-17:
I think - based on the the alpha tester thread this is now fully resolved?  If not, please post and the forum and we can talk there until we figure out what the bug might be.

closed 2020-06-17 by geeksville