meshtastic/firmware #182: Add named remote attributes, remote settings and GPIO control.

geeksville, 2020-06-14:
from @Xanuman.  

my summary:

omg - I like this idea so much.  For a few reasons:

* I've been wanting to clean up the internal structure of the device code with a very clear delineation between the mesh vs the other system code vs the GUI app that understands text/GPS messages.
* That cleanup would make it easier to have the concept 'it is easy to link in your own small code as an app on the mesh node and you can do what you want with hardware and send/receive messages (using the regular 'arduino' api)
* Your idea would be an ideal first other 'co-resident app'.  And it is pretty clean and easy to do
* @Anelito's comment about remote settings makes me think we should generalize this into "every node can have a set of named attributes that can be read/written by other nodes".  Initial usage of this named attribute concept would be for remote settings management, and this new remote-gpio module.

Slight restatement for the idea (for the github bug I'm about to make):

* Make a 'remote-gpio-module' (and it will use the new 'remote attribute API')
* Let the android/python clients have a mechanism to list, view and change named attributes
* The gpio-module would publish all of the free GPIOs on a board as named attributes available for setting, reading and direction control
* This new module could be a great small example for how others could write/add new modules of their own

Alas - I'm trying to stay super focused on getting 1.0 finished (and stable :sweat_smile:) with the original planned set of features.  So I don't think I'll have time to work on this until 1.0 'ships' (a few weeks?).  But I'm super eager to make this happen (and it sounds fun). If no one does it before then, I'll happily do so.

geeksville, 2020-06-14:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/extension-of-the-functionality-of-devices-meshtastic/411/2

Professr, 2020-06-25:
I'll work on cleaning up / compartmentalizing the GUI some, it's already most of the way there.  I'd been planning to make GUI updates delta and/or event triggered to reduce CPU load, and pare down its contact points with the rest of the app

geeksville, 2020-06-25:
that sounds awesome.

Professr, 2020-06-28:
[test branch](https://github.com/Professr/Meshtastic-device/tree/issue%23182)
Here's what I've got so far.  The screen runs at 1fps in idle mode and is drive by observations instead of the DebugInfo assignment in main.cpp's loop.
I've abstracted the main status types from the various modules that handle collecting / updating the data, so
* screen.h/cpp observes a single instance each of PowerStatusHandler, GPSStatusHandler, and NodeStatusHandler
* main.cpp tells those three handler instances to observe the appropriate modules during setup()
* I moved the power handling code out of main-esp32.cpp and into power.h and Power.cpp (not sure why the case difference, meh) so it could be a proper class.  Since it's a class it can extend PeriodicTask, so I set it to read the power status every 20 seconds (it would never read the power status unless there was an AXP interrupt, as far as I can tell, so the battery indicator wasn't updating except after a wake event or power/USB/battery disconnect/reconnect)
* For testing, there's a heartbeat pixel in the upper left corner of the screen so you can see when it redraws

This way you could have GPSStatusHandler observe multiple GPS modules.  When we need to support multiple channels, NodeStatusHandler can observe and integrate multiple NodeDBs, etc.  The gps.h class could probably be replaced with the status handler; it was doing most of that job already.  IDK if this is how you want to do it, but regardless it's a good idea to homogenize the way the three main status-generating modules communicate with the UI and any other parts that need to consume data from them.

Professr, 2020-06-28:
Oh, and the status handlers don't pass on observed events unless the data _changes_, though it'd be easy enough to add another observable that forwards all events if for some reason the raw feed was needed.

geeksville, 2020-06-28:
@Professr  that sounds really great.   When do you wanna do a PR for it?  btw - I received a message from @grcasanova who wanted to clean up namespaces and the thread/timer stuff.  So far it seems like you are both messing with mostly separate stuff, but I wanted to mention it so that no-one has to go through merge hell. ;-)

Professr, 2020-06-29:
TBH I'm really out of my depth with some of the C++ concepts involved in Observers and the more complicated parts of class instance and pointer persistence.  My OO experience is with much weaker typing and automatic garbage collection, so I've got a nagging feeling that I'm missing something important.  I can send a PR, but it'd be really great to have a sanity check from someone else before it gets merged in.

geeksville, 2020-06-29:
FWIW - I read it and it looked okay to me.  And to some extent I used Observers/Observables in a couple of cases where just a callback function would have sufficed because I already had them and thought "in the future there might be multiple clients who want to be notified when this happens"

I used to (10 yrs ago) write **lots** of C++ code but until this project I hadn't written any sizeable amount of C++ in a while.  Most of my recent experience is C/python/kotlin/scala/javascript.  So it might be that the existing C++ code wasn't ideal anyways.  IMO your changes look like good progress compared to what was there.

Professr, 2020-06-29:
Cool :)  I always prefer to fully understand the changes I'm making, but at least this is a good step.  Good practice too.

chron0, 2020-08-10:
oh yes, i only searched the issues for "sensor" and nothing came up - this is the right place. As example user stories:

Example 1: Motion Detection (event driven)

I have two nodes, one is my communication node connected to my device. The other is a node with added PIR/RADAR GPIO (ideally interrupt driven for power saving) serving to extend the mesh and also publishing motion events which I can subscribe to on my mobile device thru the other node. Initial configuration would ideally be done thru some "device config" in the android app before dropping the motion detection node.

Example 2:  Radiation/Air Quality/Temperature/Humidity/Pressure Measurements (periodic)

In this example I have a solar powered node that is equipped with a couple of sensors periodically (frequency adjustable) taking samples and publishes it. N users can subscribe to this data to either see them directly in the app or have some prometheus/influxdb "proxy" using the python API to subscribe and gather the data and then storing it.

geeksville, 2020-09-27:
As part of this feature, on ESP32 targets add a "temp" GPIO that you can read by using: https://circuits4you.com/2019/01/01/esp32-internal-temperature-sensor-example/#:~:text=ESP32%20has%20on%20chip%20temperature,on%20most%20of%20the%20ESP32.

geeksville, 2020-12-11:
this is now mostly implemented: https://meshtastic.discourse.group/t/on-device-api-now-available-feedback-requested/1872

mc-hamster, 2021-03-27:
Closing this as done.

We now have GPIO control (it's early but it works) and an admin channel for remote device control.

closed 2021-03-27 by mc-hamster