meshtastic/firmware #187: GPS is bad message on OLED

feh123, 2020-06-16:
Hi

I have a tbeam running 0.7.6 beta. I have tried updating today but continue to get gps is bad message and no position icon on mapbox. I am running the app with 0.7.76.

Is it possible to update to a device version with the latest commits? Thanks.

geeksville, 2020-06-16:
Two things to check:

* Confirm that you have the TBEAM ROM installed (not TTGO-LORA32 - that is a different board)
* Take the unit outside and set it on the ground, it can need quite some minutes for the GPS to get is initial lock and GPS almanac download.  Thereafter the GPS keeps that data in a battery backed up RAM so it should acquire a lock more quickly.  You'll know it has a lock when the red led by the GPS starts blinking.

geeksville, 2020-06-16:
If that doesn't work, please connect to the board with a serial terminal at 921600 and post the messages the board prints out into this bug and I can probably advise.

feh123, 2020-06-16:
Hi 
I can for sure make errors but the Lilygo order was a T-beam and I have successfully updated from 0.6.2 or so. I have tried it outside and that did not fix it. I attach the putty log file. 
I should say that the red light is flashing next to the gps. Any advice very much appreciated.
[putty.log](https://github.com/meshtastic/Meshtastic-device/files/4788261/putty.log)

feh123, 2020-06-17:
Hi tried going through the back versions - 0.7.5 also gave "gps bad" but 0.6.8 has worked and my app (0.7.78) is displaying the correct location for my device. I attach the putty log file for 0.6.8.
I will try a few more versions.
[putty 17-Jun-v0.6.8.log](https://github.com/meshtastic/Meshtastic-device/files/4791165/putty.17-Jun-v0.6.8.log)

feh123, 2020-06-17:
Hi I have checked the versions and 0.7.4 is working fine, 0.7.5 gives gps bad.

Much relief - I was beginning to think I had faulty devices!

geeksville, 2020-06-17:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/alpha-tester-thread-no-new-tests-needed-please-upgrade-to-release-version/298/70

cedarknoll, 2020-06-17:
I too am having this bug. (I assume it is the same bug anyway) It started with 0.7.5, I have three tbeams they are all currently on 0.7.6. I have had all three of them running outside all morning (several hours). They do have a blinking red light, but they all say "GPS bad" on display. I will attach a log from one of them.
[gps_bug.txt](https://github.com/meshtastic/Meshtastic-device/files/4794799/gps_bug.txt)

anyway hope that is helpful.

geeksville, 2020-06-17:
@cedarknoll super useful thanks - I'll fix this later today.

geeksville, 2020-06-17:
Ok - I found the problem, it seems to be a race condition in the Sparkfun_ublox library and apparently now that we are more multithreaded it shows up.  Alas, I won't be fixing this tonight but it is now my #1 bug :-) and I'll fix it tomorrow (probably).

geeksville, 2020-06-18:
oops - as it turns out I was using their API in a dumb way.  fxing today.

commit 698fcdad (2020-06-18): Merge pull request #194 from geeksville/dev

Fix #187, Ublox GPS fixes, see below:
commit cd903dce (2020-06-18): Fix #187, Ublox GPS fixes, see below:

This was a good one. Two problems
1) We've apparently always been using hte sparkfun API the wrong way, but
but we mostly got lucky.
2) Changing to use the API correctly (asyncronously) exposed a bug in
the library - fixed in a seperate commit
closed 2020-06-18 by geeksville