meshtastic/firmware #197: Show RX signal level as a percent score

geeksville, 2020-06-18:
Currently we are showing raw received signal levels.  In screen.cpp change that to scale based on the following roughly measured values to something that looks like a % value betwen 0 and 100:

> - rx signal measurements -3 marginal, -9 bad, 10 great, -10 means almost unusable. So scale this into % signal strength.

mnantel, 2020-06-22:
SNR is logarithmic, right?
Also, I have seen values of +11 reported - I have not managed to test the lowest possible value.

Professr, 2020-06-22:
Yes, I should have made that exponential (sigh).  I've occasionally seen 11 too if the units are right by each other, but in practical terms anything over 10 is pretty ideal

Professr, 2020-06-22:
Check my math here, but SNR is logarithmic with respect to 0 so we have to calculate the log of its absolute value and then return its sign.  If our range from "bad" to "good" is -10dB to 10dB, we can expect the log value to be -1 to 1.  Anything less than -1 is just as useless as -1, and anything above 1 is probably in unusually-ideal environments in close proximity, so I'm clamping the scaled offset to the 0-100 range:

`float snrLinear = log(abs(node->snr)) * (node->snr / abs(node->snr));`
`snprintf(signalStr, sizeof(signalStr), "Signal: %d%%", clamp((int)((snrLinear + 1.0) * 50.0), 0, 100));`

mnantel, 2020-06-22:
I am not much of a mathematician nor good enough a developer to critique :) I was merely passing an observation as I considered how I would have tackled this. No doubt this makes more sense - I’ll test it out later on !

Professr, 2020-06-22:
A SNR of 10 vs. 9.75 is an 11% drop going by an exponential scale.  I wonder how the inverse square law comes into play here, could the relationship between dB and *distance* be somewhat linear?  I guess it all depends on what we're trying to measure here; the underlying data might not have a straightforward relationship to a practical percentage :\

Professr, 2020-06-22:
The tbeam radio looks like it supports reading RSSI, which would be much easier to convert to a percentage.  I don't know if that's supported across other chipsets though.

closed 2020-06-23 by geeksville