meshtastic/firmware #198: When we send position updates to other nodes, we should also send it to the phone

geeksville, 2020-06-18:
So that if the phone is connected for a long time, it has a chance for its nodeinfos to get updated.  Currently it doesn't look like those messages are being put into the ToPhone queue.

aHVzY2g, 2020-06-24:
Could you elaborate on this issue a bit more? If I understand it right it is related to [MeshService::sendOurPosition](https://github.com/meshtastic/Meshtastic-device/blob/9061b3d8c3e23812d8c8d8467bc11e7f524c0714/src/mesh/MeshService.cpp#L269) ?

geeksville, 2020-06-24:
oh yes. whenever the node either broadcasts its position (very occasionally - once every 15 min) or some other node asks this node to send its position (we will unicast a reply in this case).  This is great, but our rule for "what things should go in the 'forward to phone queue'" is currently only 'stuff that was sent TO this node'.  That's not really ideal, because whenever we send out a position update others we should also have that same message provided to the local phone so it will have current map position for the local user.

I can elaborate more later.

aHVzY2g, 2020-06-24:
Ah, got it. I will dig in tomorrow, doesn't sound too hard to fix

geeksville, 2021-04-01:
fixed long ago

closed 2021-04-01 by geeksville