meshtastic/firmware #199: Give infos about GPS fix quality

kalon33, 2020-06-18:
Would it be possible to indicate some data about GPS fix quality in the app (nb of satellites, hdop, vdop...) and onto the screen (number of fixed satellites along with the ok?). That would be helpful to determine how precise and trustworthy our position is. During my tests, I sometimes had a bad quality fix that was pretty far away from my home, but I knew that because I know where my home is...

Professr, 2020-06-20:
I've put together a graphical GPS display while playing with the code, and I'm using the PDOP guide from wikipedia for accuracy.  The UBlox GPS module returns PDOP * 100, so I'm just scaling it back down and mapping it to "Ideal", "Excellent", "Good", etc.  Would that be what you're looking for?

geeksville, 2020-06-21:
That would be perfect!

Even as an initial PR before the graphics you could change the current display "GPS ok" string to instead be 0-100% instead.  If you need any help with sending in a PR I'm happy to help (or google 'github how to send in a pull request' and I bet there is a nice article)

geeksville, 2020-06-21:
Fixed by @Professr.  Yay!

closed 2020-06-21 by geeksville

kalon33, 2020-06-23:
@Professr @geeksville thanks for the implementation! That's what I was looking for yes. Would it also be possible to indicate the number of satellites used for positioning (eventually /number of satellites in view)?

Professr, 2020-06-23:
We have access to that data, but there's only so much screen real estate...  If we keep the stats in a single header bar, I think there's only room for either DOP or # of satellites, unless you can think of a way to display both pieces of data in the same space and keep it intuitive / readable

kalon33, 2020-07-02:
@Professr under the "good" or "bad" word in GPS section, would it be possible to indicate nb fixed sats/nb sats in view? There's seems to have a little room for it

kalon33, 2020-07-02:
maybe with a sat logo?

Professr, 2020-07-02:
I've been thinking about the ambiguity inherent in the standard labels for DOP categories, and I think it'll be much better to use a strength indicator.  That'll free up some space for some sort of satellite stats, will have to take a crack at coming up with something user-friendly

Professr, 2020-07-03:
The GPS libs don't support sats-in-view (well, NEMA exposes a raw satellite array, but it doesn't say whether it's in-view or just more details about the sats used in the fix).  I was able to integrate the number of sats used for the fix, though I'm still curious what that info really buys us.  Until there's something more crucial that'd bump it out of the UI space, though, it's fun to look at :)

kalon33, 2020-07-09:
For me it indicates some kind of fix robustness. And in my mind, the more satellites you have, the more precise your position is, complementary to DOP info.