meshtastic/firmware #203: Singletons for tracking gps and node status

Professr, 2020-06-20:
PowerStatus is maintained throughout the app as a singleton for display purposes, and it seems logical to do the same for GPS and NodeDB for consistency's sake.  I have it working, assuming it's alright to add the relevant namespace/structs to GPS.h and NodeDB.h respectively.  Thoughts?

geeksville, 2020-06-21:
I see where you are coming from and I think that's a great for GPS (But I though there was a gps singleton pointer already?).  But eventually I'd like to be able to run multiple nodes in one process (for simulation mesh purposes), so I'm in favor of keeping nodedb not a singleton (because each 'node' needs its own copy of the DB)

Professr, 2020-06-21:
I was thinking more of a status struct that wouldn't change the way the node db object worked.  For example, in my test code the only change is adding 
`    nodesStatus.nodesOnline = getNumOnlineNodes();
    nodesStatus.nodesTotal = getNumNodes();`
to the part where a node gets added / times out.  That way, the screen code doesn't have to call the more expensive methods to show user stats, it just has to pull from the nodesStatus singleton.

The GPS class definitely functions that way, but I wasn't certain whether the way it's instantiated made it as available as PowerStatus.  Maybe I'm missing the picture, I just thought it looked right to have 
`// Global power status singleton
meshtastic::PowerStatus powerStatus;

// Global GPS status singleton
meshtastic::GPSStatus gpsStatus;

// Global nodes status singleton
meshtastic::NodesStatus nodesStatus;
`
right at the top of main.cpp

geeksville, 2020-06-22:
ooh I see.  Yes that sounds great!

Professr, 2020-06-22:
Okay, I'll see if I can do that without changing much.  If it doesn't end up being worth it, no biggie :)

PR #237 (2020-06-29): Switch main display to event-driven model, abstract the three main status info categories, work on power.h
This changes a lot of files, so please feel free to suggest refactors or tweaks especially around the right/wrong way to be using Observables.
- The screen now runs at 1fps when idle, and it uses notifications from three `CallbackObserver`s to drive status updates.
- I added `Status.h` which provides a framework for maintaining data, subscribing to a data producer, and notifying data consumers of changes.  I made `PowerStatus.h`, `GPSStatus.h`, and `NodeStatus.h` for tracking power, gps,  [... cut]
closed 2020-06-29 by geeksville