meshtastic/firmware #205: GPS pin and Protocol Documentation

Aggebitter, 2020-06-21:
I would love to have the pin numbering used for Neo 6M on a Heltec Wifi 32 LoRa and what protocol settings for that common GPS module

Source says pin 36, 37 @ 9600, it does not work for me
Cheers Agge

geeksville, 2020-06-21:
hmm - that should be fine according to https://github.com/meshtastic/Meshtastic-device/commit/1c6092c43001d2c12a93698c67f7f480df7e84d8

@zombodotcom perhaps has extra advice?

geeksville, 2020-06-21:
And of course whatever pins you need to hook to your 6m to give it power etc...

zombodotcom, 2020-06-22:
I tried on two devices and worked fine. Make sure that your tx and rx are switches. Your tx from the chip should go to the rx pin . Rx to tx. I think. 

It's opposite. I'll reupload if that doesn't work for you.

Aggebitter, 2020-06-22:
I just made a check in U-center on the protocol settings for my 6M
Protocol in: 0+1+2 - UBX+NEMA+RTCM2
Protocol out:0+1 - UBX+NEMA
Bauderate: 9600
Databits: 8
Stopbits: 1
Parity: none
Bit Order: LSB first

The 6M fires up and gets a 3D fix in U-center, but on the heltec I got "GPS bad" but my "fixed" LED on the GPS blinks some times in a irregular matter and not as an stable PPS as it does in U-center.

Any known protocol settings ?

zombodotcom, 2020-06-22:
Make sure that you're downloading the most recent branch. and that your configuration file actually has the definitions in there. 
[image]

also set heltec  in platform.ini
[image]

Aggebitter, 2020-06-22:
@zombodotcom I just used the stable bin file for Heltec as I do all my "arduino" programing in Visual Micro and VS-community edition

zombodotcom, 2020-06-22:
It takes a second but I have No GPS, Then Ideal GPS, then Good GPS,  TX from the chip to the device is to pin 36

Aggebitter, 2020-06-22:
I just swapped the TX from my module between the Heltec and my USB serial device connected to U-center. The GPS works perfect. Averaging with more then 10 satellites. So it is not a module issue.
Even powering the 6M from the USB serial with a stable fix and cold resetting the HELTEC, still no good GPS :-). Change the Heltec still no GPS..

zombodotcom, 2020-06-22:
Just used two separate heltecs and wired them up. Uploaded the most current branch. They both have good gps.

I don't know what to tell ya.

Aggebitter, 2020-06-22:
Do you have U-center ? 
U-blox software for config GPS modules ?

zombodotcom, 2020-06-22:
Yeah it's a little finicky. It's also using tinygps++ which I used in my other code. It takes a second to get started but I tried it again today on both modules and it says gps excellent

aHVzY2g, 2020-06-24:
@Aggebitter for the testing program your Heltec as Serial Passthrough, if it doesn't show NEMA messages you probably have bad wiring.

Aggebitter, 2020-06-25:
@zombodotcom Could you check what protocols your 6M module has enabled with U-center? As mine has an history as GPS for my drone they are reprogrammed as I mentioned earlier. 
Mine is:
Protocol in: 0+1+2 - UBX+NEMA+RTCM2
Protocol out:0+1 - UBX+NEMA
Baudrate: 9600
Databits: 8
Stopbits: 1
Parity: none
Bit Order: LSB first

@aHVzY2g It's not an wire issue. 
What I can see in the source is that it first tries to program the GPS module with sending UBX configuration to the 6M module. If that fails it swaps to NEMA. Without my logic analyser I haven't been able to check what the 6M module response is yet. But I think there might be an initial setting on my 6M that make either don't take the settings sent from the HELTEC or the response is wrong as my modules are sending UBX+NEMA.

aHVzY2g, 2020-06-25:
@Aggebitter if you did program your Heltec as Serial Passthrough, please post the Serial output( and please use Pastebin for that). If you haven't done this so far, use this sketch to do so.

```
#include <Arduino.h>

void setup() {
  Serial.begin(9600);
  Serial2.begin(9600,SERIAL_8N1,36,37);
}

void loop() {
  if (Serial2.available()) {     // If anything comes in Serial2 (pins 36 & 37)
    Serial.write(Serial2.read());   // read it and send it out Serial (USB)
  }
}
```

Do not use your serial adapter for that.

aHVzY2g, 2020-06-25:
All right there two different things I can think of that may help you. 
- remove the onboard coin cell of your module if it has one. That should clear any custom settings stored in battery-backup RAM, back to stock.
- manually reconfigure the module to the default configuration. You can check 1.15.1 Boot-time configuration [here ](https://www.u-blox.com/sites/default/files/products/documents/NEO-6_DataSheet_(GPS.G6-HW-09005).pdf), the default settings are in bold.

FYI intrest this is what U-Center reports for my BN-180
[image]

geeksville, 2020-07-08:
ok -since this seems to be not a problem with any of the supported boards.  I'm going to close this bug just to keep the workqueue clear.  Feel free to still discuss.

closed 2020-07-08 by geeksville