meshtastic/firmware #211: Longer term - support for one-time-pad over shortlink?

Professr, 2020-06-22:
One of my pet topics is one-time-pad encryption.  Since node-to-node messages are currently only encrypted with the channel key, they could be exposed to anyone else on the channel who intercepts them, right?  The concept of using a short link like wifi or bluetooth when devices are in close proximity could solve the major problem with one-time-pads (distributing the pads themselves).  Either the phone app or the device itself could automatically generate and store a predefined amount of pad with each user it finds itself in shortlink range of, and then use that pad for messages directed at each specific user.  Storage considerations would only make this useful for text, but it is unbreakable and requires virtually no CPU overhead.  When the pad runs out, it would fail over to the channel key and report the change in security to the user.

geeksville, 2020-06-24:
ooh that does sound like an interesting idea.  And since the devices are already using BLE or wifi anyways for the protocol portion only a new (small) BLE/wifi client would be needed.  It could talk to the 'server' device using the existing FromRadio/ToRadio protocol.  Adding some series of 'exchange pads' packets.  The user could either click to say "yes, let that device that wants to exchange pads mine do so' or more likely 'if we both have common channel keys, treat that other device as 'trusted' and always update our supply of OTP when we happen to see each other'

Professr, 2020-06-25:
It'd be nice if we could use it to bypass the traffic overhead of AES (OTP is byte for byte), but I don't know if that's possible without leaking metadata in the headers :\  I figured the process would be automatic, since I imagine every user<->user connection would get its own channel key when they added each other as contacts or whatever.  I'm thinking salted message text for short messages, random chunked transfer for longer ones to reduce metadata exposure (I bet that could be applied to AES-encrypted messages too)... `#brainstorming`

closed 2022-03-07 by garthvh